analogdevicesinc/libiio #1027: libusb discovery fails on FreeBSD

yurivict, 2023-08-11:
In 0.25 you've added pkgconfig discovery for libusb and a hard failure when pkgconfig fails to find it.

On FreeBSD libusb comes from the base and isn't discoverable by pkgconfig.

Could you please allow for such situation and instead of hard-failing when it isn't found use these lines from the older version:
```
	#Handle FreeBSD libusb and Linux libusb-1.0 libraries
	find_library(LIBUSB_LIBRARIES NAMES usb-1.0 usb)
	find_path(LIBUSB_INCLUDE_DIR libusb.h PATH_SUFFIXES libusb-1.0)
```

?

rgetz, 2023-08-13:
That looks like it was from:

https://github.com/analogdevicesinc/libiio/commit/26a5e101ee4d3867bd1c6a480f97ae954ae311aa

Maybe @pcercuei has an idea...

Also - since that has come up more than once (sorry) - maybe @ccraluca can look at [adding FreeBSD](https://learn.microsoft.com/en-us/azure/virtual-machines/linux/freebsd-intro-on-azure) to the CI infrastructure when she is back, so we can catch this stuff before a release.

-Robin

pcercuei, 2023-08-14:
Do you know why the alternative code path fails? Because it should still work, the fallback code didn't change in that commit.

rgetz, 2023-08-14:
@pcercuei : I don't think the fallback code ever worked for FreeBSD. FreeBSD uses a path of FreeBSD `libusb` not `libusb-1.0` like Linux does.

In FreeBSD - there is this : `/lib/libusb/libusb.h`

I think that causes the new CMake : `find_path(LIBUSB_INCLUDE_DIR libusb.h PATH_SUFFIXES libusb-1.0)` to fail.

rgetz, 2023-08-14:
Can reproduce:
```
- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
CMake Error at CMakeLists.txt:159 (message):
  Unable to find libusb-1.0 dependency.

  If you want to disable the USB backend, set WITH_USB_BACKEND=OFF.
```

Since the problem is that pkg config was missing, I did:
```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -146,15 +146,16 @@ if (WITH_USB_BACKEND)
        find_package(PkgConfig)
        if (PkgConfig_FOUND)
                pkg_check_modules(LIBUSB libusb-1.0)
-               if (NOT LIBUSB_FOUND)
-                       #Handle FreeBSD libusb and Linux libusb-1.0 libraries
-                       find_library(LIBUSB_LIBRARIES NAMES usb-1.0 usb)
-                       find_path(LIBUSB_INCLUDE_DIR libusb.h PATH_SUFFIXES libusb-1.0)
-               else()
+               if (LIBUSB_FOUND)
                        set(LIBUSB_LIBRARIES ${LIBUSB_LINK_LIBRARIES})
                        set(LIBUSB_INCLUDE_DIR ${LIBUSB_INCLUDE_DIRS})
                endif()
        endif()
+       if (NOT LIBUSB_LIBRARIES OR NOT LIBUSB_INCLUDE_DIR)
+               #Handle library path differences between FreeBSD (libusb) and Linux (libusb-1.0)
+               find_library(LIBUSB_LIBRARIES NAMES usb-1.0 usb)
+               find_path(LIBUSB_INCLUDE_DIR libusb.h PATH_SUFFIXES libusb-1.0 libusb)
+       endif()
        if (NOT LIBUSB_LIBRARIES OR NOT LIBUSB_INCLUDE_DIR)
                message(SEND_ERROR "Unable to find libusb-1.0 dependency.\n"
                        "If you want to disable the USB backend, set WITH_USB_BACKEND=OFF.")
```

This works even when pkgconf is installed, since it still returns `null` for things, and enters the `NOT LIBUSB_LIBRARIES OR NOT LIBUSB_INCLUDE_DIR` test...

Things build properly after that, and even work...

```
rgetz@qemu:~/libiio/build $ sudo ./tests/iio_info -S
Password:
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.3.5]
rgetz@qemu:~/libiio/build $
```

which is a question for @yurivict : In the existing CMake, we get:
```
- Features enabled : xml network dns-sd avahi ipv6 serial usb utils man-utils
-- Features disabled: zstd bonjour local local-mmap hwmon examples iiod udev-rule c++-bindings doc man python-bindings c#-bindings
                                                                        ^^^^^^^^^
```

The `udev` rule is turned off - since : 
```
if (WITH_USB_BACKEND AND CMAKE_SYSTEM_NAME MATCHES "^Linux")
```

should that be turned on for freeBSD too?

Thanks
-Robin

rgetz, 2023-08-15:
@tfcollins - I found : https://github.com/marketplace/actions/freebsd-vm

rgetz, 2023-08-17:
looks like the `devd` will take a little more poking...

`udev` on linux returns vendor/product ID without leading `0x` indicators, but `devd` adds them - so if we scan with the leading `0x` - it breaks without a minor tweak.... 

Just trying to see how minor I can make it - ok - it's pretty minor, but it is a real bug, not just something specific to FreeBSD. :(

```
diff --git a/scan.c b/scan.c
index 74010cc9..63aa9677 100644
--- a/scan.c
+++ b/scan.c
@@ -173,7 +173,8 @@ struct iio_scan_context * iio_create_scan_context(
                 * usb=vid:pid */
                for (ptr = strstr(ctx->backendopts, "usb="); ptr;
                     ptr = strstr(ptr, "usb=")) {
-                       ptr += sizeof("usb=");
+                       /* 1 for the null char we don't want */
+                       ptr += sizeof("usb=") - 1;
                        strtoul(ptr, &ptr2, 16);
 
                        /* The USB backend will take care of errors */
```

then it works : 
```
rgetz@qemu:~/libiio/build $ sudo ./tests/iio_info -S usb=0x0456:0xb673
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.3.5]
rgetz@qemu:~/libiio/build $ sudo ./tests/iio_info -S usb=0456:b673
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.3.5]
rgetz@qemu:~/libiio/build $ 
```

previously we were eating the first char, (un)luckily ADI's VID = 0456; or we would have found it sooner.

rgetz, 2023-08-17:
and the `devd` rule (equivilent to the `udev` rule) would be:

```
"/etc/devd/libiio.conf" 9L, 253B                                                                                                                              8,102-109     All
# $FreeBSD$
#

notify 0 {
        match "system"          "USB";
        match "subsystem"       "DEVICE";
        match "type"            "ATTACH";
        action "/home/rgetz/libiio/build/tests/iio_info -S usb=$vendor:$product && chmod 666 /dev/$cdev";
};
```

Anyone can take them from here - or I can make a PR tomorrow evening sometime.

pcercuei, 2023-08-17:
What about:
```
/home/rgetz/libiio/build/tests/iio_info -S usb=${vendor:2:6}:${product:2:6} && chmod 666 /dev/$cdev
```

rgetz, 2023-08-17:
Its still a bug that needs to get fixed. We over truncate the string...

`usb=0456:b673` turns into `456:b673` so if you had: `usb=1456:b673` it still turns into `456:b673` 

that's a bug.

pcercuei, 2023-08-17:
Thanksfully this bug is most likely already fixed in the dev branch, since the scan API changed completely there.

rgetz, 2023-08-19:
yes - this bug does not exist in the dev branch:

```
rgetz@brain:~/github/libiio-mainline/build$ ./tests/iio_info -S usb=0456:b673
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.10.5]
rgetz@brain:~/github/libiio-mainline/build$ ./tests/iio_info -S usb=0x0456:0xb673
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.10.5]
rgetz@brain:~/github/libiio-mainline/build$ ./tests/iio_info --version
iio_info version: 1.0 (git tag:d09a8ac6)
Libiio version: 1.0 (git tag: d09a8ac6) backends: local xml ip usb
rgetz@brain:~/github/libiio-mainline/build$ 
```

commit b475feaa (2023-08-30): CMake: Handle discovering libusb without pkg-config

Don't require pkg-config for libusb discovery. If pkg-config is not
found, use the standard find_library / find_path approach.

Fixes #1027.
commit 53d92394 (2023-08-30): CMake: Handle discovering libusb without pkg-config

Don't require pkg-config for libusb discovery. If pkg-config is not
found, use the standard find_library / find_path approach.

Fixes #1027.
PR #1043 (2023-08-31): CMake: Handle discovering libusb without pkg-config
Don't require pkg-config for libusb discovery. If pkg-config is not found, use the standard find_library / find_path approach.

Fixes #1027.
commit 6b565571 (2023-08-31): CMake: Handle discovering libusb without pkg-config

Don't require pkg-config for libusb discovery. If pkg-config is not
found, use the standard find_library / find_path approach.

Fixes #1027.
closed 2023-08-31 by pcercuei