analogdevicesinc/libiio #1030: Streaming 122.88 MHz @ 12 bit over USB 3

cedivad, 2023-08-15:
Hello, and thank you for a great project!

I understand this is not supported today, but I've started working on your code with my custom hardware (a new SDR) to try and achieve this objective. Data needs to be packed down to 3 bytes per samples for this to work (~350 MBps), which is not a supported format, but that is a reasonable fix.

After looking at the code and testing both versions of the software, I've come to the conclusion that this project is not optimised for concurrency (for example, a single, blocking, read or write thread is used when sending data over ethernet or usb, rather than trying to parallelise the operation with multiple streams).

My assumption is that given the older generation ARM cores those FPGAs usually come with, focus has been on single-threaded performance (like zerocopy) rather than exploiting what more recent CPUs can do (think RPI4).

Am I completely wrong? I saw that one of the published objectives for 1.0 was to enable streaming with full USB 3.0 bandwidth, but my tests have been terrible and I can't find a single benchmark showing anywhere close to saturation on remote interfaces.

Many thanks!

pcercuei, 2023-08-17:
Hi @cedivad,

Have a look at what's in the "dev" branch. It will be merged into "master" really soon (in the upcoming days). It is the development branch for the upcoming Libiio v1.0, so you will notice that the API changed (see also: https://github.com/analogdevicesinc/libiio/wiki/libiio_0_to_1).

It uses a different, asynchronous network/USB protocol, which means that you can enqueue multiple "blocks" sequentially without having to wait for each one to complete.

It will be completed by the new DMABUF API (https://github.com/analogdevicesinc/libiio/pull/928), which will allow zero-copy from IIO to USB. This PR is waiting for some kernel bits to be merged upstream first, and some work on the FPGA side (we need a "scatter-gather" feature in the DMA IP block).

cedivad, 2023-08-17:
Hello @pcercuei, and thank you for your answer. I already tried what I thought was libiio1.0, but used one of your `pcercuei/*` branches, because I saw they had more activity than `dev`. I didn't see a difference in performance, probably because I was using them wrong!

Can you confirm that the new branch can come close to saturating USB3 with the correct setup? I moved on to writing my own drivers since then, but if the performance is there I'll be happy to give `libiio` another go!

pcercuei, 2023-08-17:
No, you'll need more than that.

You need what's in PR #928, plus the IIO DMABUF kernel patch (https://patchwork.kernel.org/project/linux-media/list/?series=736449), plus the FunctionFS DMABUF kernel patch (https://patchwork.kernel.org/project/linux-media/list/?series=732667).

Finally, you need a scatter-gather capable DMA IP core with the corresponding driver changes. These are TBD.

cedivad, 2023-08-17:
Thank you for the pointers, I have a better idea of what's involved now. None of this seems too hard, but I should probably hold off from trying to port it to my platform for the moment.

To stay in-topic, is there any plan for packing 12 bits samples into 3 bytes rather than the 4 that are used today? Doing so requires changing the FPGA core, IIO subsystem and more, so adoption would be very limited, but even something as simple as supporting the datatype inside libiio (host side) would let people running their custom iiod server (like myself) full compatibility with the released project binaries.

pcercuei, 2023-08-18:
I don't think you would need to change anything in Libiio or IIOD, since it just streams the data as-is, and it's up to your application to decode it.

Maybe `iio_channel_convert` or `iio_channel_convert_inverse` would need to change, but that's about it.

cedivad, 2023-08-18:
I should have spent more time looking into it, but the issue I saw was that after setting a device to 12 storage/real bits IIOD is going to assume a sample size of 4 bytes, while it should be 3. My assumption here was that this wouldn't have worked across the chain all the way to the user output. 

Anyways, it's ok, I'll try and send you a PR when I can get a prototype working. Closing this for now, hope I can open it again when I've made some progress :-)

closed 2023-08-18 by cedivad