analogdevicesinc/libiio #1036: libiio 1.0 scan not operating the same as 0.25

rgetz, 2023-08-23:
in libiio 1.0 (master) - I don't know if this is on purpose or not, but this is different than it use to be...

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr --version
iio_attr version: 1.0 (git tag:58cf11d)
Libiio version: 1.0 (git tag: 58cf11d) backends: local xml ip usb
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -u usb: -v -S
WARNING: No backend found for scan string 'usb:'
No IIO context found.
rgetz@brain:~/github/libiio/build$ echo $?
1
```

In the past - if `-S` was given, it would do the scan, and ignore the `-u` 

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -u usb: -v -S
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10447376de0b000213002900fafa9908ec [usb:3.12.5]
rgetz@brain:~/github/libiio/build$ echo $?
0
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr --version
iio_attr version: 0.25 (git tag:a015afec)
Libiio version: 0.25 (git tag: a015afe) backends: local xml ip usb
```

Not sure this is bad - but it is different. 

As a min, It should either error with a meaning error message, or do the scan, and ignore ...
(since there actually is a context at `usb:`):
```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -u usb: -C
IIO context with 15 attributes:
hw_model: Analog Devices PlutoSDR Rev.C (Z7010-AD9361)
hw_model_variant: 1
hw_serial: 10447376de0b000213002900fafa9908ec
fw_version: v0.35
ad9361-phy,xo_correction: 39999959
ad9361-phy,model: ad9361
local,kernel: 5.10.0-98231-g9dfba10b795d
uri: usb:3.12.5
usb,vendor: Analog Devices Inc.
usb,product: PlutoSDR (ADALM-PLUTO)
usb,serial: 10447376de0b000213002900fafa9908ec
usb,idVendor: 0456
usb,idProduct: b673
usb,release: 2.0
usb,libusb: 1.0.24.11584
```

pcercuei, 2023-08-24:
Well, I don't really consider that a bug, since you are passing bogus arguments...

rgetz, 2023-08-24:
anything on the command line is not "bogus" - it is user error. I agree with that -- but you need to protect users from themselves, and point them in the right direction.

Printing out random errors, is the bug.

```
WARNING: No backend found for scan string 'usb:'
```

`usb:` isn't the scan string (it's the uri) so something is going wrong with the way things are parsed. That's the bug.

-Robin

pcercuei, 2023-08-24:
It's just iio_common.c having a single `arg` variable that's set by both -S and -u handlers, those should be made mutually exclusive.

pcercuei, 2023-08-24:
Should be fixed by #1039.

pcercuei, 2023-09-27:
Closing as the issue is assumed to have been addressed.

closed 2023-09-27 by pcercuei