analogdevicesinc/libiio #1042: Error when opening tx buffer with libiio 0.25 on host and 1.0 on device

catkira, 2023-08-30:
When I try to open a buffer with `buffer = iio_device_create_buffer(dev, 7680, false);` I get this error
```
Open unlocked: -12
```

pcercuei, 2023-08-30:
-12 is ENOMEM, that's strange.
Does the same `buffer = iio_device_create_buffer(dev, 7680, false);` works if you run it on the device directly?

> It does not show the git hash when I do `iio_info -V` on the device, is that intended (tag v1.0 does not exist yet)?

It should show the git hash as long as you built it from the git repository.

catkira, 2023-08-30:
I get the same error when I try it with libiio 1.0 on host.
I cannot easily run the code on device, because it is embedded in a GUI app.
My code looks like this:
```c++
#pragma once
#include <iostream>
#include <chrono>
extern "C"{
    #include "srsran/phy/utils/vector.h"
}

#include "tx_streamer.h"

Tx_streamer::Tx_streamer()
{
    ctx = nullptr;
}

void Tx_streamer::start()
{
    open_iio_buffer();
    stop_worker_thread = false;
    worker_thread = std::make_unique<std::thread>(&Tx_streamer::worker, this);
    struct sched_param sp;
    sp.sched_priority = 100;
    pthread_setschedparam(worker_thread->native_handle(), SCHED_OTHER,  &sp);
}

// void Tx_streamer::stop()
// {
//     close_iio_buffer();
//     stop_worker_thread = true;
//     if (worker_thread)
//     {
//         std::cout << "waiting for worker_thread to join..." << std::endl;
//         worker_thread->join();
//     }
// }

void Tx_streamer::subframe_callback(std::unique_ptr<std::array<std::complex<float>, SF_SAMPLE_SIZE>> &&subframe)
{
    if (buffer != nullptr)
        subframe_in_queue.push(std::move(subframe));
}

void Tx_streamer::init_iio_device(std::string ip)
{
    if (ctx != nullptr)
        return;

    // Initialize libiio context
    printf("create iio context\n");
#if LIBIIO_VERSION_MAJOR == 1
    ctx = iio_create_context(NULL, (std::string("ip:") + ip).c_str());
#else
    ctx = iio_create_context_from_uri((std::string("ip:") + ip).c_str());
#endif
    if (!ctx) {
        std::cerr << "Failed to create libiio context" << std::endl;
        return;
    }
}

void Tx_streamer::close_iio_device()
{
    iio_context_destroy(ctx);
    printf("iio_context_destroyed\n");
    ctx = nullptr;
}

[... 205 lines cut ...]
            auto t_now = std::chrono::high_resolution_clock::now();
            auto duration = std::chrono::duration<float>(t_now - t_1);
            t_1 = std::chrono::high_resolution_clock::now();
            printf("tx_streamer: %f frames/s, idle %f %%\n", 1000 / duration.count(), duration_wait.count() / duration.count() * 100);
            duration = std::chrono::milliseconds::zero();
            duration_wait = std::chrono::milliseconds::zero();
        }

    }

#if LIBIIO_VERSION_MAJOR == 1
    for (auto &block : blocks){
        iio_block_destroy(block);
        block = nullptr;
    }
#endif

}

void Tx_streamer::worker()
{
    write_iio_buffer();
}

```

catkira, 2023-08-30:
`iio_rwdev -u ip:192.168.137.2 -b 7680 cf-ad9361-dds-core-lpc voltage0 -B -w`
is working

pcercuei, 2023-08-30:
It sounds like a valid "out of memory" error then. I think if you are using the "linux-mmap" code with ADI's kernel, the blocks created will be backed by contiguous memory provided by the Contiguous Memory Allocator (CMA) in Linux. You might need to add a dedicated memory area for the CMA if it's not already the case, or increase its size.

pcercuei, 2023-08-30:
Those look normal to me, if you already have 2 blocks, you need to create a 3rd one, hence the +1.

catkira, 2023-08-30:
With libiio 0.25 I can set the kernel buffers count to 20 without problems, using the command `iio_device_set_kernel_buffers_count()`, blocksize was the same.

nunojsa, 2023-08-30:
What I meant was that if you want to just create 2 blocks you end up creating 3... Because on the second time you run the code you set `req.count = 2` so you create two blocks on top of the existing one. But I might be missing something on the libiio side.

pcercuei, 2023-08-30:
Looking at the kernel code, it seems to only create the additional blocks requested - so if you requested 2 before, and now request 3, it will only allocate one more. At least that's what I understand from the kernel code.

nunojsa, 2023-08-30:
I don't think so... If you request 3, then you'll create 3. I looked for about 5 minutes (so I might be wrokng), but it seems to me that in v0.25 it works precisely because we just create all the blocks in [enable_high_speed](https://github.com/analogdevicesinc/libiio/blob/v0.25/local.c#L814). Note in that case we set `req.count = nb_blocks;`

Look also at this [line](https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/buffer/industrialio-buffer-dma.c#L618).

pcercuei, 2023-08-30:
Yep you're right. I'll fix it tomorrow.

pcercuei, 2023-08-31:
@catkira can you confirm that PR #1044 fixes your issue?

catkira, 2023-08-31:
it's working. Thanks for fixing it!

commit 9931d471 (2023-08-31): local-mmap: Fix block allocation

The algorithm for allocating new blocks was broken, as it was requesting
an allocation of (nb_blocks + 1) blocks instead of a single block.

There is also no need to check the return value of req.count, as we are
allocating a single block now; if the allocation fails the ioctl() will
return an error code.

Fixes #1042.
PR #1044 (2023-08-31): local-mmap: Fix block allocation
The algorithm for allocating new blocks was broken, as it was requesting an allocation of (nb_blocks + 1) blocks instead of a single block.

Fixes #1042.
commit 9a82e375 (2023-08-31): local-mmap: Fix block allocation

The algorithm for allocating new blocks was broken, as it was requesting
an allocation of (nb_blocks + 1) blocks instead of a single block.

There is also no need to check the return value of req.count, as we are
allocating a single block now; if the allocation fails the ioctl() will
return an error code.

Fixes #1042.
closed 2023-08-31 by catkira