analogdevicesinc/libiio #1045: CMAKE fails on Raspberry Pi3A+

ingframin, 2023-08-31:
Hello, As per title, Cmake fails to generate the make files on Raspberry Pi3A+.
I installed Raspberry Pi OS Lite 64bit released 03/05/2023.
I attached the log.

Output of Cmake:
-- cmake version: 3.18.4
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
-- Performing Test HAS_WSHADOW - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.24
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found Git: /usr/bin/git (found version "2.30.2") 
CMake Error at CMakeLists.txt:428 (message):
  Unable to find libzstd dependency.

  If you want to disable ZSTD compression support, set WITH_ZSTD=OFF.

-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Found LibXml2: /usr/lib/aarch64-linux-gnu/libxml2.so (found version "2.9.10") 
-- Found BISON: /usr/bin/bison (found version "3.7.5") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
-- Features enabled : xml zstd network dns-sd avahi ipv6 local local-dmabuf local-mmap hwmon usb utils iiod udev-rule iiod-serial iiod-aio iiod-usb man-utils
-- Features disabled: bonjour serial examples modules usb-dynamic network-dynamic serial-dynamic iiod-systemd iiod-sysvinit iiod-upstart doc man python-bindings c#-bindings c++-bindings
-- LOG_LEVEL set to "Info"
-- Configuring incomplete, errors occurred!
See also "/home/stratosense/libiio/build/CMakeFiles/CMakeOutput.log".

[CMakeOutput.log](https://github.com/analogdevicesinc/libiio/files/12484544/CMakeOutput.log)

pcercuei, 2023-08-31:
Well this is pretty self-explanatory, "Unable to find libzstd dependency". You can disable ZSTD support with `-DWITH_ZSTD=OFF`. If you don't need IIOD or the network / USB / serial backends, you can disable them all together and you won't need ZSTD at all.

ingframin, 2023-09-06:
Thanks, I am going to try again. I found it weird because 0.25 builds without issues.

commit 300c8c23 (2023-12-07): README: add libzstd-dev to the list of prerequisites

In ./README_BUILD.md we keep track of Prerequisites, now for libiio 1.x

should resolve #1045