analogdevicesinc/libiio #1046: lio.h file not found

avalerobiot, 2023-08-31:
I am trying to compile libiio following the instructions in the README_BUILD file and I get the same error both in MacOs and in Linux Centos:

Fixup Current symbolic link
[ 56%] Built target iio-compat
[ 59%] Linking C shared library iio.framework/iio
[ 59%] Built target iio
[ 65%] Building CXX object bindings/cpp/CMakeFiles/iiopp-enum.dir/examples/iiopp-enum.cpp.o
[ 65%] Building C object tests/CMakeFiles/iio_tests_helper.dir/gen_code.c.o
[ 65%] Building C object tests/CMakeFiles/iio_tests_helper.dir/iio_common.c.o
In file included from /Users/albertovalero/libiio/bindings/cpp/examples/iiopp-enum.cpp:12:
libiio/bindings/cpp/./iiopp.h:17:10: fatal error: 'iio.h' file not found
#include <iio.h>
         ^~~~~~~
[ 68%] Linking C static library libiio_tests_helper.a
[ 68%] Built target iio_tests_helper
[ 72%] Building C object tests/CMakeFiles/iio_attr.dir/iio_attr.c.o
[ 72%] Building C object tests/CMakeFiles/iio_info.dir/iio_info.c.o
[ 75%] Building C object tests/CMakeFiles/iio_stresstest.dir/iio_stresstest.c.o
[ 77%] Building C object tests/CMakeFiles/iio_rwdev.dir/iio_rwdev.c.o
[ 79%] Building C object tests/CMakeFiles/iio_genxml.dir/iio_genxml.c.o
[ 81%] Building C object tests/CMakeFiles/iio_reg.dir/iio_reg.c.o
[ 84%] Linking C executable iio_genxml
[ 86%] Linking C executable iio_info
[ 88%] Linking C executable iio_reg
[ 90%] Linking C executable iio_rwdev

Am I missing something?
Thanks!

rgetz, 2023-12-01:
I don't know if @sternmull wants to have a look at this one?

pcercuei, 2023-12-01:
I'm working on it.

illiw99, 2023-12-21:
Any updates on this? Encountered the same problem!

pcercuei, 2023-12-21:
Still not done - it gets pushed down the priority list constantly.

pcercuei, 2024-01-12:
The updated C++ bindings are in the main branch now, closing.

closed 2024-01-12 by pcercuei