analogdevicesinc/libiio #1048: wrong error checks for result from iio_create_context()

catkira, 2023-09-03:
I did error checks for the result of iio_create_context like here
https://github.com/analogdevicesinc/libiio/blob/9a82e3757208630886f6aff25f723d9ca95a9501/tests/iio_stresstest.c#L252C1-L268C4

However the problem was, that in case of an error a negate value gets return which was not caught by this 'if(!ctx)'.

I changed my code to 
```c
    ctx = iio_create_context(NULL, (std::string("ip:") + ip).c_str());
    int err = iio_err(ctx);
    if (err) {
        std::cerr << "Failed to create libiio context" << std::endl;
        ctx = nullptr;
        return;
    }
```
and then it worked correctly.

pcercuei, 2023-09-04:
Thanks, can you send a PR? Since you already have the fix 😬

dNechita, 2025-01-20:
This has been fixed. See the PR referenced above.

closed 2025-01-20 by dNechita