analogdevicesinc/libiio #1050: `__cnst` incorrectly used, subtle bugs possible

marcusmueller, 2023-09-12:
As figured out in https://github.com/gnuradio/gnuradio/pull/6823#discussion_r1323755582 :

> ```C
> __api __check_ret __cnst const struct iio_data_format * iio_channel_get_data_format(
> 		const struct iio_channel *chn);
> ```
> 
> `__cnst` expands to `__attribute__((const))`, but that tells GCC "If you pass in the same pointer (not the identical object pointed to!) twice, you get the same result"; but that's not actually the case, because a pointer to the same address might point to a different object next invocation, so the data format might be different. See [GCC's warning about use with pointers, as well](https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#index-const-function-attribute).

`iio_channel_get_data_format` is, in GCC attribute parlance, *((pure))*, but not *((const))*.

pcercuei, 2023-09-13:
I disagree. The use of "const" is correct here.
- `iio_channel_get_data_format` does not examine the data pointed to at any moment. In fact, it never deferences the pointer.
- If you pass the same pointer (and as you said, not the identical object pointed to), you will always get the same result, which is a pointer to the pointed channel's data format structure. If the underlying channel object changed between two calls, but their address is the same, then the address of their data structure will be the same.

marcusmueller, 2023-09-13:
aaaah, I missed the fact that it's not returning `chn->format`, but `&(chn->format)`, d'oh. sorry for the noise!

closed 2023-09-13 by marcusmueller