analogdevicesinc/libiio #1053: Can get SIGFPE if device has bad scan info.

dlech, 2023-09-13:
While developing a kernel driver, I ran into a crash due to `SIGFPE` here: https://github.com/analogdevicesinc/libiio/blob/e2ba5d485bc7890c914c24a32e606a8d47b1bf4b/device.c#L466

This was due to `scan_type` not being configured in the driver yet and therefore `length` here in libiio was 0 and caused a divide by 0 error.

It would be better to check for this and create a runtime error instead of crashing.

Searching for `% length`, it looks like there are several places where this might need to be checked: https://github.com/search?q=repo%3Aanalogdevicesinc%2Flibiio+%2F%25+length%2F&type=code

rgetz, 2023-09-13:
Maybe the thing to do is check and fail early? 

in https://github.com/analogdevicesinc/libiio/blob/main/local.c#L763-L835

when we start to decode the scan_type?

if it's bad - we should return a failure there...

pcercuei, 2023-09-14:
Well, I disagree. Libiio does not have to handle every case of a driver behaving against the specification.

nunojsa, 2023-09-15:
This feels like it should be handled by the IIO core when the device is probing...

pcercuei, 2024-01-12:
I'm going to close this issue as "won't fix", Libiio will only work properly if the device drivers follow the specifications.

closed 2024-01-12 by pcercuei