analogdevicesinc/libiio #1061: Empty attribute results into EIO

darkmoon32, 2023-10-04:
When I use `iio_info` I can see in the log IO errors when accessing the devices' trigger. This is because the **current_trigger** is empty and reading the file results in reading zero bytes. Is this the desired behavior? When I change [this line](https://github.com/analogdevicesinc/libiio/blob/92d6a35f3d8d721cda7d6fe664b435311dd368b4/local.c#L717) to accept zero as a valid result, the error disappears but the library thinks that the device does not have a trigger at all.

The iio_info output:
```
        iio:device2: lis3dh (buffer capable)
                4 channels found:
                        accel_x:  (input, index: 0, format: le:s12/16>>4)
                        3 channel-specific attributes found:
                                attr  0: raw value: -72
                                attr  1: scale value: 0.009806
                                attr  2: scale_available value: 0.009806 0.019613 0.039226 0.117679
                        accel_y:  (input, index: 1, format: le:s12/16>>4)
                        3 channel-specific attributes found:
                                attr  0: raw value: 4
                                attr  1: scale value: 0.009806
                                attr  2: scale_available value: 0.009806 0.019613 0.039226 0.117679
                        accel_z:  (input, index: 2, format: le:s12/16>>4)
                        3 channel-specific attributes found:
                                attr  0: raw value: -980
                                attr  1: scale value: 0.009806
                                attr  2: scale_available value: 0.009806 0.019613 0.039226 0.117679
                        timestamp:  (input, index: 3, format: le:S64/64>>0)
                4 device-specific attributes found:
                                attr  0: current_timestamp_clock value: realtime

                                attr  1: mount_matrix value: 1, 0, 0; 0, 1, 0; 0, 0, 1
                                attr  2: sampling_frequency value: 1
                                attr  3: sampling_frequency_available value: 1 10 25 50 100 200 400 1600
                2 buffer-specific attributes found:
                                attr  0: data_available value: 0
                                attr  1: watermark value: 1
                1 debug attributes found:
                                debug attr  0: direct_reg_access value: 0x0
ERROR: checking for trigger : Input/output error (5)
        iio:device3: 48003000.adc:adc@0 (buffer capable)
                2 channels found:
                        voltage9:  (input, index: 0, format: le:u12/16>>0)
                        3 channel-specific attributes found:
                                attr  0: offset value: 0
                                attr  1: raw value: 1294
                                attr  2: scale value: 0.610351562
                        voltage19:  (input, index: 1, format: le:u12/16>>0)
                        3 channel-specific attributes found:
                                attr  0: offset value: 0
                                attr  1: raw value: 1297
                                attr  2: scale value: 0.610351562
                4 device-specific attributes found:
                                attr  0: oversampling_ratio value: 1
                                attr  1: oversampling_ratio_available value: 1 2 4 8 16 32 64 128 256 512 1024
                                attr  2: trigger_polarity value: rising-edge
                                attr  3: trigger_polarity_available value: rising-edge falling-edge both-edges
                2 buffer-specific attributes found:
                                attr  0: data_available value: 0
                                attr  1: watermark value: 1
                7 debug attributes found:
                                debug attr  0: lincalfact6 value: 512
                                debug attr  1: lincalfact5 value: 537395712
                                debug attr  2: lincalfact4 value: 536347136
                                debug attr  3: lincalfact3 value: 537395712
                                debug attr  4: lincalfact2 value: 536347135
                                debug attr  5: lincalfact1 value: 538443263
                                debug attr  6: direct_reg_access value: 0x100A
ERROR: checking for trigger : Input/output error (5)
```

pcercuei, 2023-10-05:
Can you confirm that https://github.com/analogdevicesinc/libiio/pull/1065 fixes your problem?

(The commit is on top of the `main` branch which has a lot of API changes, if you are still using v0.25, you can cherry-pick the fix on top of the `libiio-v0` branch to test locally)

darkmoon32, 2023-10-05:
Yes, this resolves the issue for `libiio-v0` branch.

pcercuei, 2023-10-05:
Thanks. I'll cherry-pick the commit to `libiio-v0` as soon as the PR is merged.

PR #1065 (2023-10-10): local: Fix reading of empty attributes
This partially reverts 3fa8e97b. This commit is from 2014, I have no idea why we wanted -EIO for empty attributes. Maybe I thought that this would never happen.

Fixes #1061.
commit bacf0874 (2023-10-10): local: Fix reading of empty attributes

This partially reverts 3fa8e97b. This commit is from 2014, I have no
idea why we wanted -EIO for empty attributes. Maybe I thought that this
would never happen.

Fixes #1061.
closed 2023-10-10 by pcercuei