analogdevicesinc/libiio #1082: Unable to install on Raspi 3B

Tac97, 2023-11-08:
I have followed the process for building and installing IIO on a Raspi running Debian Bullseye 64 bit. When I run 'iio_info' I get:
'''
iio_info version: 1.0 (git tag:877b9b0b)
Libiio version: 1.0 (git tag: 877b9b0b) backends: local network usb xml
Backend version: 1.0 (git tag: 877b9b0b)
Backend description string: Linux pi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
IIO context has 2 attributes:
	local,kernel: 6.1.21-v8+
	uri: local:
IIO context has 2 devices:
	hwmon0: cpu_thermal
		1 channels found:
			temp1:  (input)
			2 channel-specific attributes found:
				attr  0: crit value: 110000
				attr  1: input value: 51540
		No trigger on this device
	hwmon1: rpi_volt
		1 channels found:
			in0:  (input)
			1 channel-specific attributes found:
				attr  0: lcrit_alarm value: 0
		No trigger on this device
'''
but when I try to install the python bindings from the 'setup.py.cmakein' file I get:
'''
/usr/lib/python3/dist-packages/setuptools/dist.py:475: UserWarning: The version specified ('${VERSION}') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
running install
Traceback (most recent call last):
  File "/home/test/libiio/bindings/python/setup.py.cmakein", line 71, in _check_libiio_installed
    raise OSError
OSError
'''
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/test/libiio/bindings/python/setup.py.cmakein", line 100, in <module>
    setup(**config)
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/test/libiio/bindings/python/setup.py.cmakein", line 43, in run
    self._check_libiio_installed()
  File "/home/test/libiio/bindings/python/setup.py.cmakein", line 77, in _check_libiio_installed
    raise Exception(msg)
Exception: The libiio library could not be found.
            libiio needs to be installed first before the python bindings.
            The latest release can be found on GitHub:
            https://github.com/analogdevicesinc/libiio/releases

I have a Python script tested on  a linux machine that uses pylibiio and pyadi-iio that i need to run on the Raspi so any advice is appreciated.

tfcollins, 2023-11-08:
I see you are install libiio v1.0 which is in an unstable stabte. If you are using pyadi-iio you need to use libiio v0.25 or older. Then just you pip to install pylibiio

Tac97, 2023-11-10:
Thanks for the help. I've rolled back to Libiio version 0.25 and built it with the CPP_BINDINGS on, but still get the same errors when trying to run SETUP.PY.CMAKEIN or running a Python script that imports IIO and ADI. Is there something simple I'm missing here? Do you mean something other than building with CPP bindings when you talk about the c library?

Tac97, 2023-11-10:
I have fixed the problem. I used a fresh Raspberry Pi OS image (The full 64 bit version), installed the Libiio v0.25 Ubuntu arm64 version and pip installed `pylibiio` and `pyadi-iio`

closed 2023-11-10 by Tac97