analogdevicesinc/libiio #1094: Simulink IP Core not showing in available context

Beauxrel, 2023-12-06:
How can I include my ip core "mwipcore" into the iio available context? I add it to the device tree and run the script to create a .strip file, but it doesnt show up in iio_info.

Beauxrel, 2023-12-06:
[image]

tfcollins, 2023-12-06:
What kernel are you using? and branch?

Beauxrel, 2023-12-06:
https://github.com/analogdevicesinc/linux/tree/2021_R1

tfcollins, 2023-12-06:
I really have no idea what that driver does that matches the "mwipcore" compat string. Its maintained by MathWorks and is pretty ancient.

There is a newer driver from the MathWorks kernel that does use that compat string but is very different.

Beauxrel, 2023-12-06:
What do you recommend I do?

Im wondering if I can add this to my defconfig found here: [adi_zynqmp_defconfig](https://github.com/analogdevicesinc/linux/blob/b52f687e9deb4e36875b55c6a511dff3c8c35c81/arch/arm64/configs/adi_zynqmp_defconfig#L152)

I've been hacking the zynq stuff where I can since that's what the workflow is actually looking for

`CONFIG_MATHWORKS_IP_CORE=y
CONFIG_MWIPCORE=y
CONFIG_MWIPCORE_DMA_STREAMING=y
CONFIG_MWIPCORE_IIO_STREAMING=y
CONFIG_MWIPCORE_IIO_MM=y
CONFIG_MWIPCORE_IIO_SHAREDMEM=y
CONFIG_MATHWORKS_GENERIC_OF=y`

Beauxrel, 2023-12-06:
Also, can you link what youre referring to.

tfcollins, 2023-12-06:
https://github.com/mathworks/xilinx-linux/blob/master/drivers/misc/mathworks/mathworks_ipcore.c

tfcollins, 2023-12-15:
Closing as this is not related to libiio itself

closed 2023-12-15 by tfcollins