analogdevicesinc/libiio #1104: RX buffer issues libiio V1

tfcollins, 2023-12-14:
Similar issue to #1097 ,

Using looped back ramp signals data is not contiguous or is but contains some bias/scaling.

See test code here: https://github.com/analogdevicesinc/libiio/blob/tfcollins/testing/tests/hardware/test_ad9364.c
This tests will create a ramp signal in cyclic mode with digital loopback enable. Then on the RX side it will verify each received buffer contains this ramp somewhere and is contiguous.

Receiving from IIO-Scope is correct but with the libiio V1 there seem to be a bit offset. The data exceeds expected values and tends to changed with some power of two value.

pcercuei, 2023-12-15:
That's a bug in your code. Your `p_end` pointer points to the end of the last *TX* block, and you don't update it to point to the end of the current *RX* block.

tfcollins, 2023-12-15:
Thanks for checking this Paul. That fixed it

closed 2023-12-15 by tfcollins