analogdevicesinc/libiio #1106: Clarification v1.0 API

raccoman, 2023-12-19:
I am trying to create a simple application that reads a buffer of 1024 samples continuously. I found the example ad9361-iiostream.c that would seem to do the trick. The only problem is that I don't understand how the new iio_device_create_buffer works. It accepts an index which from wiki would seem correct to leave 0 but then accepts a mask which I honestly don't understand what it is for. Also iio_device_get_sample_size() always returns 4 but I don't know how to change it and if it needs to be changed. Then iio_buffer_create_stream I understand that from the internal buffer takes tot blocks (1 parameter) as big as samples_count (2 paramenter) continuously without having to call refill?

catkira, 2023-12-20:
yes you can leave the index at 0. The mask stores which channels are enabled, but you don't manipulate the mask directly, instead you do it with calls to `iio_channel_enable()`, ie:
```c++
    unsigned int nb_channels = iio_device_get_channels_count(dev);
    mask = iio_create_channels_mask(nb_channels);
    struct iio_channel *ch = iio_device_get_channel(dev, 0);
    iio_channel_enable(ch, mask);
```
In this example only channel 0 is enabled.
The sample size cannot be changed, this value is set by the device driver and depends on the number of enabled channels.

pcercuei, 2023-12-21:
`iio_buffer_create_stream` will create `nb_blocks` iio_block objects of `samples_count` samples each, that will be enqueued to the device. Each refill will dequeue *one* of the blocks and re-enqueue the previous one.

raccoman, 2023-12-21:
Thanks for the answers, so taking this example:

```c
    tuner->rx_stream = iio_buffer_create_stream(tuner->rx_buf, 4, 1024 * 1024);
    if (iio_err(tuner->rx_stream)) {
        logger_error("tuner", "tuner_init() failed: iio_buffer_create_stream() failed");
        return -1;
    }

    size_t rx_sample_size = iio_device_get_sample_size(tuner->rx_stream_dev, tuner->rx_mask);

    const struct iio_block *rxblock;
    ssize_t nrx = 0;

    while (1) {
        int16_t *p_dat, *p_end;
        ptrdiff_t p_inc;

        rxblock = iio_stream_get_next_block(tuner->rx_stream);
        if (iio_err(rxblock)) {
            logger_error("tuner", "tuner_init() failed: iio_stream_get_next_block() failed");
            break;
        }

        /* READ: Get pointers to RX buf and read IQ from RX buf port 0 */
        p_inc = rx_sample_size;
        p_end = iio_block_end(rxblock);
        for (p_dat = iio_block_first(rxblock, tuner->rx0_i); p_dat < p_end;
             p_dat += p_inc / sizeof(*p_dat)) {
            /* Example: swap I and Q */
            int16_t i = p_dat[0];
            int16_t q = p_dat[1];
        }

        nrx += 1024 * 1024 / rx_sample_size;
        //logger_debug("tuner", "\tRX %8.2f MSmp\n", nrx / 1e6);
    }

```

When should I call buffer refill? Also `iio_device_get_sample_size` what does returns? It seems that it's used to jump between blocks, so it could be just block size? (1024 * 1024)?

pcercuei, 2023-12-21:
It is, yes. Internally, `iio_stream_get_next_block` will call `iio_block_enqueue` on the previous block, and `iio_block_dequeue` on the next block, which are the functions that handle the streaming of samples from/to the hardware.

raccoman, 2023-12-22:
From a performance point of view, what is the relationship between number of blocks and number of samples per block? For example, what is different from 1 block of 1024*4 versus 4 blocks of 1024 samples each?

raccoman, 2024-01-04:
Also I would like to known the proper way to destroy all the objects allocated. Currently this is my code:
```c
int tuner_destroy(t_tuner *tuner) {

    pthread_cancel(tuner->thread_rx);
    logger_debug("tuner", "pthread_cancel() success");

    iio_channel_disable(tuner->rx0_i, tuner->rx_mask);
    iio_channel_disable(tuner->rx0_q, tuner->rx_mask);
    logger_debug("tuner", "iio_channel_disable() success");

    iio_stream_destroy(tuner->rx_stream);
    logger_debug("tuner", "iio_stream_destroy() success");

    iio_buffer_destroy(tuner->rx_buf);
    logger_debug("tuner", "iio_buffer_destroy() success");

    iio_channels_mask_destroy(tuner->rx_mask);
    logger_debug("tuner", "iio_channels_mask_destroy() success");

    iio_context_destroy(tuner->ctx);
    logger_debug("tuner", "iio_context_destroy() success");
    return 0;
}
```
The problem with this is that often segfault or freeze on buffer destroy.

Last but not least, during my application routine I need to be able to switch to another frequency / configuration. In this case, disabling the channels, destroying the stream and the buffer is enough to change the configuration e start it again right?

pcercuei, 2024-01-04:
I think I may have already fixed your crash, try to cherry-pick https://github.com/analogdevicesinc/libiio/pull/1035/commits/f4f6053a06e8b084d66894f63f4de9697d921c27 (it's in the "new DMABUF API" pull-request) and see if it works better.

pcercuei, 2024-01-04:
As for the way you destroy objects - it looks OK to me, no problem there.

raccoman, 2024-01-04:
> As for the way you destroy objects - it looks OK to me, no problem there.

Thanks, I'll try. For the previous question about performance?

pcercuei, 2024-01-04:
Switching to a different frequency should just be a write to a iio_attr, you shouldn't have to tear down the stream and buffer and recreate them.

As for your previous question, bigger blocks will generally give you a slightly better performance, because the per-block overhead is smaller relative to the block size. But using big blocks will increase the latency so it's a tradeoff. Libiio v1.0 will also do a much better job than v0.25 with small (a few KB) blocks so the difference in performance is not as huge as it used to be.

raccoman, 2024-01-04:
> Switching to a different frequency should just be a write to a iio_attr, you shouldn't have to tear down the stream and buffer and recreate them.
>
> As for your previous question, bigger blocks will [...]

Thanks! Unfortunately, even with the other version from the cherry pick, it doesn't work. It freezes when it has to destroy the buffer

raccoman, 2024-01-04:
Here is some debug results:

[image]
When calling line 193:`ops->free_buffer(buf->pdata);` it fails.

[image]