analogdevicesinc/libiio #1109: cannot recover when iio_block_dequeue() timeouts

catkira, 2023-12-20:
I think we briefly talked about this already.

When a call of `iio_block_dequeue()` with nonblock set to false timeouts, the return error is `ETIMEDOUT`. This is fine. What I would expect is that the next call to `iio_block_dequeue()` then tries again for the timeout period that has been set with `iio_context_set_timeout()`. However what happens is that all calls to `iio_block_dequeue()` now immediately return with error code `EPERM` even when new blocks are available to dequeue. 

Can this be changed so that a read timeout is recoverable without having to tear down the whole connection and start everything from new? The current behavior prevents my data link application from doing fast reconnects in case of a connection loss.

catkira, 2023-12-20:
@pcercuei <3

pcercuei, 2023-12-20:
@catkira Try my `pcercuei/survive-timeout` branch. Completely experimental, completely untested, of course.

(Note that this branch will generate a non-working IIOD so don't use it on the board.)

catkira, 2023-12-20:
I will try it tomorrow.

What happens with the data inside a partly filled block when a timeout happens? Is the data discarded or kept inside the partly filled block and returned with the next dequeue command when the block is full?

catkira, 2023-12-21:
@pcercuei it doesn't work yet, after the first timeout all following dequeue commands return immediately with `ETIMEDOUT`.

pcercuei, 2023-12-21:
The code I added *should* support network disconnections when they happen between commands, if it is in the process of transferring the data, recovery is a whole lot more difficult...

catkira, 2023-12-21:
In my case there is no disconnection of the network between the ad9361 and client iio-device, it's just a disconnection of the wireless connection from the ad936x to the other ad936x. In that case the axi_dmac buffer does not get filled anymore from my custom hdl code.

catkira, 2023-12-21:
For both is fine, if the data of the partially filled block gets discarded or not, because my data stream is tagged so I can detect what is old and new data.

catkira, 2023-12-21:
The following behaviour would be ok for me in case of dequeue timeout:
a) the partly filled or empty block gets discarded, and the next dequeue command will wait again for the specified timeout 
b) if a block is partly filled, the data inside that block is preserved when a timeout happens and new data is appended to it when the next dequeue command is issued

catkira, 2023-12-21:
In my case a timeout is always caused because the axi_dmac core does not deliver enough data to fill the block, _not_ because the network connection between iiod running on the ad936x and iio-client running on PC is interrupted.

catkira, 2023-12-21:
In case of a timeout and a partially filled block the dequeue command can just return with `ETIMEDOUT `, but in that case the user can check the used_bytes flag of the returned block if partial data is available. That would also be harmonious if you add support for the abort feature of the axi_dmac core with dmabuf later.

catkira, 2023-12-21:
no DMABUF for sure, I think mmap. How can I find out if its mmap or fileio?

pcercuei, 2023-12-21:
Just add a printf() to `local_dequeue_mmap_block` in local-mmap.c, you'll see if it's called ;)

catkira, 2023-12-21:
but you said I should not use iiod of this branch right?

catkira, 2023-12-21:
its a bit difficult because my device stuff is build by buildroot. But I looked at the buildroot log files and all the local-mmap stuff is enabled, so I guess it is used because it has priority over fileio right?

catkira, 2023-12-21:
this is a grep in my device build dir, WITH_LOCAL_MMAP_API=ON so I think it should use mmap.
[image]

pcercuei, 2023-12-21:
Try with this commit (in my `pcercuei/survive-timeout` branch): https://github.com/analogdevicesinc/libiio/commit/ba3dad4835e193a2ff14c2afebb4cb3caf42795f

catkira, 2023-12-21:
yes, it does not work. When I try this version on device and client, the call to `iio_create_context()` just hangs.

pcercuei, 2023-12-21:
Ok, try the `pcercuei/fix-mmap` branch on your target. I did not test timeouts (I don't really have the setup to test it) but it should allow you to re-try dequeueing the blocks.

catkira, 2023-12-21:
the branch is not there

catkira, 2023-12-21:
ok now it's there

catkira, 2023-12-21:
which rev should I use on client?

pcercuei, 2023-12-21:
I think the main branch is fine, but you need commit 1efcd1b7 (you can just cherry-pick it there).

catkira, 2023-12-21:
It's not working. iio_info gives this error
```bash
benja@benjaopti7000:~/$ iio_info -u ip:192.168.137.2
ERROR: Unable to create IIO context ip:192.168.137.2: Connection reset by peer (104)
```

pcercuei, 2023-12-21:
We're working on fixing block enqueue/dequeue here, `iio_info` does nothing of that sort ;)
Make sure you have the right versions installed (`pcercuei/fix-mmap` on the target, main + 1efcd1b on host).

catkira, 2023-12-21:
yes, but since my app could not create an iio_context I tried to run iio_info on client to see if that can create an iio_context.

catkira, 2023-12-21:
it would be a good feature if iio_info could show git hash of local and client, currently it only show git tag

pcercuei, 2023-12-21:
It shows the git tag if the commit it was built from corresponds to a tag, otherwise it shows the git hash :)

catkira, 2023-12-21:
I am trying this rev now on local
[image]
but `iio_info` on local still only shows the tags
[image]

pcercuei, 2023-12-21:
I think your problem is that Buildroot strips the .git subfolders when extracting the sources (or it downloads the release tarball from github, which does not have the .git subfolder). Libiio's CMake then cannot compute the git hash.

catkira, 2023-12-21:
the 61fdbe rev also could not connect, I tried a rev that's a bit older on local and it works again
d905a5bc1f6b7061073c95641482e3af152f98f9

catkira, 2023-12-21:
I think there are some changes unrelated to the current dequeue effort that broke some things for me

catkira, 2023-12-21:
72e8276b62371cbc1c802cc265c873d7695fd6b4 is broken, so somewhere between 72e8276b62371cbc1c802cc265c873d7695fd6b4 and aa2aa8ee06a380d9830adb4c8c5e8f6a3ccfc68e must be a problem.

pcercuei, 2023-12-21:
The problem is `72e8276b`.

pcercuei, 2023-12-21:
@catkira I force-pushed the `pcercuei/fix-mmap` branch, and rebased the changes on top of the last known good commit. And I'll have a look at that other problem later.

catkira, 2023-12-21:
after the first timeout all following calls to dequeue immediately return with `ETIMEDOUT`, maybe you forgot to reset the timeout counter somewhere?

catkira, 2023-12-21:
or maybe the timeout flag of the file descriptor needs to be cleared?

pcercuei, 2023-12-21:
> after the first timeout all following calls to dequeue immediately return with `ETIMEDOUT`, maybe you forgot to reset the timeout counter somewhere?

Can you try to reproduce it on the target directly first? Do you also get `ETIMEDOUT` when trying to dequeue a new block there?

> after a timeout, is the block dequeued or not. Should the next dequeue command be issued with the same block or the next block?

After a timeout you can consider that the dequeue operation failed, so you should re-try with the same block.

catkira, 2023-12-21:
> > after the first timeout all following calls to dequeue immediately return with `ETIMEDOUT`, maybe you forgot to reset the timeout counter somewhere?
> 
> Can you try to reproduce it on the target directly first? Do you also get `ETIMEDOUT` when trying to dequeue a new block there?

that is not easily possible, because my client app has a GUI and cannot run on the ZYNQ device.

> 
> > after a timeout, is the block dequeued or not. Should the next dequeue command be issued with the same block or the next block?
> 
> After a timeout you can consider that the dequeue operation failed, so you should re-try with the same block.

ok that is what I expected.

catkira, 2023-12-21:
Do you think the problem is that `buffer_check_ready()` does not wait after the first timeout occured?

catkira, 2023-12-21:
can you make a revision with some debug printfs that I can observe on local terminal?

pcercuei, 2023-12-21:
The behaviour you originally reported is that it would report `-EPERM` the second time you call `iio_block_dequeue`. Why do you get `-ETIMEDOUT` now? Can you try with the `pcercuei/fix-mmap` branch on the target, and the main branch on the PC?

catkira, 2023-12-21:
this is what I did, main + this extra commit on pc and fix-mmap on target

catkira, 2023-12-21:
When I try with `pcercuei/fix-mmap` on PC as well I get `Operation not permitted (1)` after the first timeout.

pcercuei, 2023-12-21:
I think there are actually two issues, one on the target side which should have been fixed with the `pcercuei/fix-mmap` branch, and another one on the PC side. With the new protocol the block enqueue/dequeue is actually one single command; `iiod_client_enqueue_block` will send the command, and `iiod_client_dequeue_block` will read the answer.

This is annoying because it means that if dequeueing times out, `iiod_client_dequeue_block` won't actually re-send the command, and that's why you get the timeout error. And you can't re-send the command anyway, since technically the block will have been enqueued on the IIOD side already and can't be enqueued a second  time.

I think I can come up with a solution, but not before the end of the year.

catkira, 2023-12-21:
ok, I will just wait then. Thanks for the effort!

catkira, 2023-12-21:
maybe it's better to dequeue the block everywhere after a timeout and then do the next dequeue command with the next block?

pcercuei, 2023-12-21:
Can you try my `pcercuei/fix-mmap` branch again? On both sides.

catkira, 2023-12-21:
now I don't get any timeout, instead I get a `Operation not permitted (1)` when a timeout should happen.

catkira, 2023-12-21:
and no timeout return from the dequeue call

catkira, 2023-12-21:
I think what happens now is that in case of a timeout it is not signalled correctly by the return value of `iio_block_dequeue()` and then the next call to `iio_block_enqueue()` fails with `Operation not permitted (1)` because the timeout block is still enqueued.

pcercuei, 2023-12-21:
Ok, now `iio_block_dequeue()` should return the timeout error.

catkira, 2023-12-21:
I only have to update client right?

catkira, 2023-12-21:
it seems to be working now!

pcercuei, 2023-12-21:
That's great!
I will have to rework the code a bit (but not this year :p) because since we send the `IIOD_OP_RETRY_DEQUEUE_BLOCK` command the moment the `iio_block_dequeue` times out, we actually start the timeout timer before the application actually calls `iio_block_dequeue` again.

pcercuei, 2024-01-09:
@catkira I reworked the code (still in the `pcercuei/fix-mmap` branch), now the `IIOD_OP_RETRY_DEQUEUE_BLOCK` is sent on the second call to `iio_block_dequeue`. Can you confirm that it still works?

catkira, 2024-01-09:
I rebased the fix-mmap branch on master before the iio_event stuff was added because my applications need to open multiple contexts locally. 
https://github.com/catkira/libiio/commits/fix-mmap-rebased/
Do you think you can make it possible to open multiple contexts locally with the new iio_event stuff? Otherwise I would have to cherry-pick everything I should test and it becomes quite messy.

catkira, 2024-01-11:
I just have to cherry-pick the 5 commits you did 2 days ago onto my rebase branch to test this right?

catkira, 2024-01-11:
or is it enough if I test the new commits on remote? Then I could use your branch directly.

pcercuei, 2024-01-11:
> I just have to cherry-pick the 5 commits you did 2 days ago onto my rebase branch to test this right?

Yes. Or test it on top of PR #1118.

catkira, 2024-01-11:
can you rebase [onto fix-mmap](https://github.com/analogdevicesinc/libiio/tree/pcercuei/fix-mmap) onto [pcercuei/local-buffer-fd](https://github.com/analogdevicesinc/libiio/tree/pcercuei/local-buffer-fd) ?

[13 more replies]

PR #1120 (2024-01-12): Better support for enqueue/dequeue errors
Given that the IIOD protocol has a single command for enqueue + dequeue, in case of an error we need to have a way to know which operation failed, so that the client can retry without having to tear down the whole IIO buffer.

Fixes #1109.
closed 2024-01-12 by pcercuei