analogdevicesinc/libiio #1110: Undefined C References in .so lib

illiw99, 2023-12-21:
Getting the following error message:
```bash
/usr/bin/ld: CMakeFiles/adis16460_example.dir/main.c.o: in function `main':
main.c:(.text+0x10): undefined reference to `iio_create_context_from_uri'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/adis16460_example.dir/build.make:104: adis16460_example] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/adis16460_example.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
```

Here is my minimum example 
```C
#include <stdio.h>
#include <iio.h>

int main(void) {

    struct iio_context *ctx;
    struct iio_device *dev0;
    
    ctx = iio_create_context_from_uri("ip:analog.local");
    
    if (!ctx) {
        perror("Unable to create IIO context");
        return -1;
    }

    dev0 = iio_context_find_device(ctx, "iio:device0");

    iio_context_destroy(ctx);

    return 0;
}
```

```CMake
cmake_minimum_required(VERSION 3.10)
project(adis16460_example C)

set(CMAKE_C_STANDARD 11)

list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR})
find_package(LibIIO 1.0 REQUIRED)

add_executable(${PROJECT_NAME} main.c)

target_include_directories(${PROJECT_NAME} PUBLIC ${LibIIO_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} PUBLIC ${LibIIO_LIBRARIES})
```

I removed the old libiio versions but still getting the error with current main branch. 

Any help is much appreciated.

pcercuei, 2023-12-21:
Well, that's an easy one. `iio_create_context_from_uri` is not in Libiio v1.0. Use `iio_create_context` instead.

closed 2024-01-09 by illiw99

IsaacSherman, 2024-08-07:
OK, stupid question, but where do I get Libiio v1.0?  All I can find are 0.x versions... I'm having the opposite problem.  When I use the CSharp bindings, I get the error that the iio_create_context is not in the dll (and iio_create_context_from_uri IS in it, based on a dumpbin).

tfcollins, 2024-08-07:
v1 is the main branch. No tags exist as it's not stable yet. Unless you need v1 features or are a developer it's recommended to use v0 releases