analogdevicesinc/libiio #1140: SIG11 in comptibility API V1.0

vvsvvsvvs, 2024-02-01:
iio_device_attr_write_bool  (compat.c) does not check the return from iio_device_find_attr(), that CAN be NULL if attr not found (it is 0, or NULL, not -1 as expected on error).
Thus, if attr was not found for a dev, there is a SIG11 issue.
So, it may be an issue for the whole API with functions returning a pointer.

vvsvvsvvs, 2024-02-01:
Running with Valgrind (current git version):
```
==390139== Command: ./iio-monitor
==390139==    
==390139== Invalid read of size 4
==390139==    at 0x486388B: iio_attr_find (attr.c:150)
==390139==    by 0x4867F08: iio_context_find_attr (context.c:511)
==390139==    by 0x4877421: dnssd_add_scan_result (dns_sd.c:115)
==390139==    by 0x4877DA0: dnssd_context_scan (dns_sd.c:309)
==390139==    by 0x4869C71: iio_scan (scan.c:122)
==390139==    by 0x10B3FB: show_contexts_screen (iio-monitor.c:263)
==390139==    by 0x10BCB1: main (iio-monitor.c:445)
==390139==  Address 0xffffffffffffffe9 is not stack'd, malloc'd or (recently) free'd
```

NOTE: sometimes code works well.

pcercuei, 2024-02-01:
The check after the `iio_context_create` is wrong, it's a pointer-encoded error - the check should actually be `if (iio_err(ctx)) { ... }`. The "ctx" pointer is therefore not valid and it cascades into the issue you pasted above.

commit 2d22c17b (2024-02-20): compat: Fix behaviour of attributes write functions

When an attribute is not found, the attributes write functions should
return -ENOENT instead of crashing badly.

Fixes #1140.
closed 2024-02-20 by mhennerich