analogdevicesinc/libiio #1142: Systemd warns about unset variables

dennis-ju, 2024-02-02:
During startup of iiod, we get the following warning:
`iiod.service: Referenced but unset environment variable evaluates to an empty string: IIOD_EXTRA_OPTS`

This is because since v254 of systemd, unset environmental variables will result in a warning (see [commit](https://github.com/systemd/systemd-stable/commit/f331434d13488425ccd8485327085d15f8f92aea)).

I could make a PR where I add this to the service file:
`Environment=IIOD_EXTRA_OPTS=''`

Would this be a good solution?

dennis-ju, 2024-02-07:
Yes, I believe that adding the line `Environment=IIOD_EXTRA_OPTS=''` would set it to an empty string while still being overridden if it is set in `/etc/default/iiod`. 

In the systemd manual (see [link](https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html)) under `EnvironmentFile=` it says "Settings from these files override settings made with Environment=."

dennis-ju, 2024-02-23:
Is there a possibility that we could add this to the legacy branch?

mhennerich, 2024-02-23:
> Is there a possibility that we could add this to the legacy branch?

Yes

mhennerich, 2024-02-23:
Lets re-open

dennis-ju, 2024-10-24:
Unfortunately the issue is not solved yet. A rogue dollar sign causes the assignment of IIOD_EXTRA_OPTS to be invalid. I'll add a new fix soon.

PR #1144 (2024-02-20): iiod: Add default env var to service file
Systemd now gives a warning when IIOD_EXTRA_OPTS is unset.

This change sets it to an empty string by default while still allowing it be overridden by /etc/default/iiod.

Fixes #1142.
commit 7ba23c8f (2024-02-20): iiod: Add default env var to service file

Systemd now gives a warning when IIOD_EXTRA_OPTS is unset.

This change sets it to an empty string by default while still allowing
it be overridden by /etc/default/iiod.

Fixes #1142.
PR #1146 (2024-03-04): iiod: Add default env var to service file
Systemd now gives a warning when IIOD_EXTRA_OPTS is unset.

This change sets it to an empty string by default while still allowing it be overridden by /etc/default/iiod.

Fixes #1142.
commit a436ad60 (2024-03-04): iiod: Add default env var to service file

Systemd now gives a warning when IIOD_EXTRA_OPTS is unset.

This change sets it to an empty string by default while still allowing
it be overridden by /etc/default/iiod.

Fixes #1142.
closed 2024-11-08 by mhennerich