analogdevicesinc/libiio #1149: what steps exactly to run ad9361-iiostream

Argonghost, 2024-03-02:
hi,

what steps exactly do i need to run the ad9361-iiostream.c example???
I cannot seem to compile normally via ``gcc`` .  I have the impression that running ad9361 streaming example requires use of CMakeLists and ``make`` etc.  My goal is to run the ad9361-iiostream example on my linux machine connected to pluto via usb.  What steps should i follow to do that? 

thanks

tfcollins, 2024-03-04:
CMake is not required. Just link against libiio 

'''bash
wget https://github.com/analogdevicesinc/libiio/blob/libiio-v0/examples/ad9361-iiostream.c
gcc -o ad9361-iiostream ad9361-iiostream.c -liio 
'''

Argonghost, 2024-03-04:
> CMake is not required. Just link against libiio 
> 
> ```bash
> wget https://github.com/analogdevicesinc/libiio/blob/libiio-v0/examples/ad9361-iiostream.c
> gcc -o ad9361-iiostream ad9361-iiostream.c -liio 
> ```

I get the error ´´segmentation fault core dumped ´´

Even despite entering my usb id

tfcollins, 2024-03-04:
Can you provide the command and full output? Have you tried to get a backtrace?

Argonghost, 2024-03-04:
> Can you provide the command and full output? Have you tried to get a backtrace?

What is backtrace??
Regarding the full command its as you provided, the full output is: 

Segmentation fault (core dumped)

tfcollins, 2024-03-07:
> What is backtrace??

This is a standard debugger output up to a failure. If you are not familiar with C debugging in Linux I would google around for gdb tutorials on basic debugging. 

 

> Regarding the full command its as you provided, the full output is:
> 
> Segmentation fault (core dumped)

This does not include the commands you used

Argonghost, 2024-03-07:
hi @tfcollins 

yes i forgot backtrace means ```gdb``` and ```run```.  on AD wiki i was finally shown how to properly define cmake to finally get it up and running .  

Thanks,

closed 2024-03-07 by Argonghost