analogdevicesinc/libiio #1162: Data conversion documentation

ahmetakman, 2024-05-06:
Hello everyone,

I am unable to understand the exact format conversion achieved with iio_channel_convert .

It says in the documentation the output goes to uint16_t but I did not get how exactly this is achieved.

Maybe more importantly, when I try to run very same configuration from MATLAB, I get consisted quite small double values whereas when I get samples and convert them from hardware format using       `iio_channel_convert(rx0_q, &converted_data, &q); `  , I get many zeros and only a few too huge numbers. 

Can you help me understand or provide a documentation that I could not reach before so that I can get the similar set of conversion using C? 

Best,

tfcollins, 2024-05-07:
Can you provide some context? What is the buffer format you have now? What device are you talking to? Have you tried the existing tools that use these functions like IIO-Scope and pyadi-iio?

tfcollins, 2024-06-21:
No response. Closing

closed 2024-06-21 by tfcollins