analogdevicesinc/libiio #1165: make error for zstd - libiio

Mas313, 2024-05-09:
Hi, 
Running make i get following error in iiod-client.c, although zstd is installed in /usr/lib/zstd  (output of which zstd). I have also copied zstd.h in the libiio folder
I have used following commands
sudo apt-get install  zstd
sudo apt install libzstd-de

output of make

iiod-client.c:975:14: error: ‘ZSTD_CONTENTSIZE_UNKNOWN’ undeclared (first use in this function)
   if (len == ZSTD_CONTENTSIZE_UNKNOWN ||
              ^
/home/masood/Downloads/libiio/iiod-client.c:975:14: note: each undeclared identifier is reported only once for each function it appears in
/home/masood/Downloads/libiio/iiod-client.c:976:14: error: ‘ZSTD_CONTENTSIZE_ERROR’ undeclared (first use in this function)
       len == ZSTD_CONTENTSIZE_ERROR) {
              ^
/home/masood/Downloads/libiio/iiod-client.c: In function ‘iiod_client_create_context_private’:
/home/masood/Downloads/libiio/iiod-client.c:1084:14: error: ‘ZSTD_CONTENTSIZE_UNKNOWN’ undeclared (first use in this function)
   if (len == ZSTD_CONTENTSIZE_UNKNOWN ||
              ^
/home/masood/Downloads/libiio/iiod-client.c:1085:14: error: ‘ZSTD_CONTENTSIZE_ERROR’ undeclared (first use in this function)
       len == ZSTD_CONTENTSIZE_ERROR) {

Any suggestions

tfcollins, 2024-05-09:
Can you provide more detail on your Linux distribution, commit used, and cmake arguments used?

tfcollins, 2024-06-21:
No response closing.

closed 2024-06-21 by tfcollins