analogdevicesinc/libiio #1174: Error when running iio_info

SrikanthPagadarai, 2024-06-04:
I grabbed the latest Azure builds for Windows and ran iio_info -s. It opens an error window with the message "The procedure entry point sp_blocking_read could not be located in the dynamic link library C:\WINDOWS\system32\libiio1.dll."

tfcollins, 2024-06-21:
A few things:
- Based on the iss file libzstd is not included but its built against it in CI
- Serial support seems broken on Windows
- Python bindings are also broken on Windows

cristina-suteu, 2024-06-26:
@tfcollins regarding the serial support , a fix was added here https://github.com/analogdevicesinc/libiio/pull/1163 , is serial support still broken after this ? 
as for the other things, i'm working on fixing them

rgetz, 2024-11-01:
I think this is fixed, and can be closed?

closed 2024-11-01 by tfcollins