analogdevicesinc/libiio #1178: Can't build 0.25 on Alpine Linux

jeffrizzo, 2024-06-19:
When trying to build 0.25 under Docker using Alpine Linux (it built in version 0.23), I got this error, so I'm reporting it as requested:

```
#11 8.002 -- Looking for pthread_setname_np
#11 8.206 -- Looking for pthread_setname_np - found
#11 8.212 -- found unknown distribution Alpine Linux, Version : VERSION_ID=3.20.0
#11 8.212 CMake Error at cmake/LinuxPackaging.cmake:35 (message):
#11 8.212   please report an error to
#11 8.212   https:\/\/github.com\/analogdevicesinc\/libiio\/issues
#11 8.212 Call Stack (most recent call first):
#11 8.212   CMakeLists.txt:650 (include)
```

tfcollins, 2024-06-21:
Can you provide your cmake command with arguments you are using? And Alpine version?

rgetz, 2024-08-26:
@jeffrizzo  This is expected - since the code doesn't support Alpine yet when building packages.

https://github.com/analogdevicesinc/libiio/blob/main/cmake/LinuxPackaging.cmake#L23-L33

Since Alpine doesn't use deb or rpm (it uses apk) - I assume you shouldn't build packages, and can just leave that off? 

Thanks
-Robin

cz172638, 2024-08-26:
https://git.alpinelinux.org/aports/tree/testing/libiio/APKBUILD

tfcollins, 2024-08-26:
@cz172638 Can we close this issue? Assuming the package works...

cz172638, 2024-08-26:
i think so, i've put link to apkbuild, so @jeffrizzo can take inspiration and eventually lobby for getting libiio package from `testing` to `community`

closed 2024-08-26 by tfcollins