analogdevicesinc/libiio #1190: C# Bindings on Linux

IsaacSherman, 2024-08-07:
Hi! I'm interested in using the C# bindings to connect to libiio from dotnet on Linux. The thing that's throwing me off right now, conceptually, is that I want to do it from Linux, but the bindings seem to be explicitly for Windows.  I want to modernize the C# bindings so they work with .NET 8 and later.  So far what I've done is I've added a .NET 8 project to the repo, changed the binding name so it would be agnostic to extension, and built with a copy of the binary I was able to download from the Releases section.  However, I run into an error: the iio_create_context entry point isn't found, and when I look in the dll, I instead find an entry point iio_create_context_from_uri as the closest option.  I've picked up from here that there is a V1.0 lurking around somewhere, but I can't find it if binaries exist.  So I assume it is what you can compile from the main branch, but I can't get cmake to build the repo on Windows.  It's missing 3 libraries, libusb, libzstd, and libxml2.  After further research, I realize libusb and libzstd are not required.  LibXml2 is required.  I'm working on finding and installing those libraries, but I'd like to know if that's even going to work at this point.  
So I have 3 questions. 
First, am I on the right track at all?
Second, assuming I'm on the right track, are there binaries I can download for the latest CI builds?  I can see there used to be some, but they don't seem to be maintained anymore.  That would save a tremendous amount of work it appears.
Finally, if there are no binaries to be had, can someone post instructions for building on windows that don't come with a warning label?  I've attempted to follow the instructions [here](https://wiki.analog.com/resources/tools-software/linux-software/building_libiio_for_windows) and have again encountered multiple errors.  I am currently trying to install LibXML2 without success (I have posted a request for help on their git as well).  

Thank you for any assistance you can provide!

tfcollins, 2024-08-07:
I'll close this as its basically the same question(s) https://github.com/analogdevicesinc/libiio/discussions/1189

Just an FYI there are no C# developers on the core teams anymore for libiio so help may be limited but we'll try our best.

With respect to versioning, please use v0.X if you are not actively targeting v1 features. Also I'm not sure if you are mixing and matching pre-built binaries with source but if you are make sure the source you are using matches the same version of which the binaries were built from.

I'm alittle confused on if you are using Linux or Windows...
- For Linux dependencies are just installed through your OS's package manager
- For Windows the dependencies can be found at the library author's site. Here is how we get them in CI https://github.com/analogdevicesinc/libiio/blob/ffbbcd184ffee64302c5f7843eb6be47b40ff585/azure-pipelines.yml#L134. Alternatively they are in the zip's for each release of libiio.

IsaacSherman, 2024-08-07:
Thanks for the quick response!  The short answer is Yes, I need to be on either/both Linux and Windows, so as long as I have the dependencies installed it shouldn't matter which platform I'm on.  I think you're right that the issue is that I have .25 binaries and 1.0 bindings.  I don't believe I need to recompile the binaries- I think I can get away with just using the old .25 bindings. 
I'll mark this closed, I have the bindings working after reading through your message and slapping myself in the forehead for missing that you had tags in your repo.  I thought I was stuck with the 1.0 bindings! I have since pulled the .25 source and built the bindings and they work.
I might do a PR making using the C# bindings a little easier.  The cmake file targets pretty old tech... .NET Framework 4, which is at least a decade old and married to Windows.  It also targets Mono, but it would be easy to retain that functionality; I wonder if it's necessary, though- dotnet is now available on Linux, and so removing the Mono requirement would reduce the complexity of the script.  I've already changed the cmake file on my machine to get it to work using the dotnet CLI without installing the tools it was looking for.  I've also added visual studio project files, but I understand if you don't want that in your repo.
If I go about doing a PR, how should I do it?  I think that the .25 needs changes (or at least a NuGet package, see below) as well as 1.0, so what would be your preferred approach? 
I would also suggest adding a NuGet package so that the bindings are available to C# developers in my position, though that would involve your CI process so I wouldn't know where to start for it.  It's pretty easy in theory- the dotnet CLI will do it with a single command, and adding the artifact to nuget.com is straightforward.  NuGet is the de facto package manager for Windows C# development- it just makes code really, really accessible.

closed 2024-08-07 by IsaacSherman