analogdevicesinc/libiio #1195: World-writable device nodes necessary?

wfrisch, 2024-09-03:
The IIO USB device nodes are set to mode 666 in `libiio.rules.cmakein`:

> SUBSYSTEM=="usb", PROGRAM=="/bin/sh -c '@CMAKE_INSTALL_FULL_BINDIR@/iio_info -S usb=%s{idVendor}:%s{idProduct} | grep %s{idVendor}:%s{idProduct}'", RESULT!="", MODE="666"

While this is easy to manage as it allows all users of a system to access IIO devices, it undermines the system's privilege separation. A more secure configuration, in my humble opinion, would be to add a system group `iio` with mode `660`.

wfrisch, 2024-09-03:
Ideally, this should be configurable via cmake so that packagers can easily set the desired group/mode for their distribution.

StefanBruens, 2024-09-03:
Preferably:

1. Change `MODE="666"` to e.g. `ENV{libiio_matched}="1"`
2. Add a rules file which actually grants the permissions:
   - `ENV{libiio_matched}=="1" TAG+="uaccess"` _or_
   - `ENV{libiio_matched}=="1" GROUP="iiod" MODE="660"`

The latter can be easily changed/disabled with a local dropin in /etc/udev/rules.d/.

StefanBruens, 2024-09-03:
Btw, the udev match rules itself has *still* several issues, see my earlier https://github.com/analogdevicesinc/libiio/issues/586#issuecomment-1372407607 .