analogdevicesinc/libiio #1199: iio_buffer_create_stream

d150kks, 2024-09-17:
when calling the iio_buffer_create_stream method in C++, the command window displays:
malloc_consolidate(): invalid chunk size
Aborted
I am using ADRV9009

tfcollins, 2024-09-18:
Can you provide more detail on your setup? What context type are you using? What commit are you building from (and commit/version of remote host if using one)?

d150kks, 2024-09-18:
I am using [8bce8b3], version 0.25 of the remote host. To run, I use a standard example but converted from C to C++. AD9009-zu11eg is connected to a PC via ethernet. 
In the same way, everything happens without errors until the stream is created.

d150kks, 2024-09-18:
I also want to add that when using the library version 0.24 on the host, the transmitter is started, but it sends the same samples, i.e. the data in the buffer is not updated, although they are loaded. How can this be fixed?

tfcollins, 2024-09-18:
We will take a look but it may take awhile.
As noted by the readme, v1 code (basically the main branch), is under heavy development and can be unstable. If you need a solution now use code from a release.

d150kks, 2024-09-19:
Understood, thanks