analogdevicesinc/libiio #1205: Cannot use libiio as a cmake subproject

myzinsky, 2024-10-05:
```
cmake_minimum_required(VERSION 3.10)
include(FetchContent)
project(qluto4)

set(CMAKE_INCLUDE_CURRENT_DIR ON)

message(STATUS "Fetching libiio from github")
FetchContent_Declare(
    libiio
    GIT_REPOSITORY https://github.com/analogdevicesinc/libiio
    GIT_TAG main
)

include_directories(build/_deps/libiio-build)

FetchContent_MakeAvailable(libiio)

```

When I want to build this project, it fails with the following error: 

```
...
[ 30%] Built target iio
[ 31%] Building C object _deps/libiio-build/CMakeFiles/iio-compat.dir/compat.c.o
[ 32%] Building C object _deps/libiio-build/CMakeFiles/iio-compat.dir/dynamic-unix.c.o
[ 34%] Linking C shared library iio.framework/iio
Fixup Current symbolic link
/bin/sh: line 0: cd: /Users/myzinsky/Programming/qluto4/build/iio.framework/Versions: No such file or directory
make[2]: *** [_deps/libiio-build/iio.framework/Versions/0.99/iio] Error 1
make[2]: *** Deleting file `_deps/libiio-build/iio.framework/Versions/0.99/iio'
make[1]: *** [_deps/libiio-build/CMakeFiles/iio-compat.dir/all] Error 2
make: *** [all] Error 2
```

So it seems that cmake places the generated framework `iio.framework` not where it actually expects it. I can find the file at:

`build/_deps/libiio-build/iio.framewok`

Seems that libiio uses [CMAKE_BINARY_DIR](https://cmake.org/cmake/help/latest/variable/CMAKE_BINARY_DIR.html) variable when create the version file: [CMakeLists.txt:374](https://github.com/analogdevicesinc/libiio/blob/main/CMakeLists.txt#L374). Most likely, the project expects given directory to refer to the project's binary directory. So that makes it impossible to build up proper projects using FetchContent. Would there be a quick solution to support this?

rgetz, 2024-10-11:
I think this is Darwin specific - Maybe @tfcollins can have a look?

tfcollins, 2024-10-11:
This seems to have been solved here https://github.com/analogdevicesinc/libad9361-iio/issues/130

rgetz, 2024-10-18:
@myzinsky does the other issue solve this one (I'm not sure - they looked seperate to me).

-Robin

myzinsky, 2024-10-18:
Somehow magically yes. However, I would prefer to use ```FetchContent_Declare``` instead of ```ExternalProject_Add```, as for many other projects.