analogdevicesinc/libiio #1206: Was there once a master branch that got renamed?

glaure, 2024-10-18:
Hi,

i just wanted to let you now that the renaming of master to main was a breaking change for a lot of yocto based firmware builds.

I fixed for me with a bappend file, but this has to be done by everyone using yocto and openembedded at a fixed released state.

Bye Gunther

nunojsa, 2024-10-21:
From what I can tell, things are fixed in upstream meta-oe. So if people use stable branches things should just work when upgrading from one release to another. Now, if some downstream hacking is being done on the recipe then, yeah, things can break (as it happened for meta-adi). 

> I fixed for me with a bappend file, but this has to be done by everyone using yocto and openembedded at a fixed released state.

The way I see it, it is already done as the rename already happened upstream. If someone is doing some downstream strange hacking (as meta-adi to get the dev branch) I would say is up to those users to fix things.

catkira, 2024-11-12:
A little inconvenience for being woke is not a big sacrifice :P

dNechita, 2025-01-20:
Closing this as the initial question has been answered.

closed 2025-01-20 by dNechita