analogdevicesinc/libiio #1211: iio_buffer_destroy() sometimes hangs when items are enqueued

catkira, 2024-11-05:
This is caused by iiod_responder_wait_done() in iiod-responder.c. The thread is probably not stopped correctly and therefore cannot join.

catkira, 2024-11-05:
The problem seems to be that iiod_rw_all() is not returning.

@pcercuei do you have a quick idea how to fix this?

catkira, 2024-11-05:
`void do_cancel(struct iiod_client_pdata *io_ctx)` in network-unix.c gets called, but it doesnt seem to abort the recv() call ^^

nunojsa, 2024-11-05:
> void do_cancel(struct iiod_client_pdata *io_ctx) in network-unix.c gets called, but it doesnt seem to abort the recv() call ^^

I guess that only matters for cancellable. But if that [flag](https://github.com/analogdevicesinc/libiio/blob/790a6c9324c775e38ab319534d97f0e3253336ff/network.c#L100) is set to true, it should do the trick. But if we are not getting into poll(2), then that cancel_fd should not matter.

But in that case MSG_DONTWAIT should be set and recv(2) should not block...

catkira, 2024-11-05:
`cancellable &= !(flags & MSG_DONTWAIT);` evaluates to true in my case

catkira, 2024-11-05:
it seems like` wait_cancellable(io_ctx, true, timeout_ms);` is blocking in my case and never returns

catkira, 2024-11-05:
something with the cancel_fd is not working

nunojsa, 2024-11-05:
Yeah, it seems there's some subtle bug in that logic... Because `do_cancel()` should do the trick

catkira, 2024-11-05:
The problem is `pfd[1].revents` is always 0 even if cancel was called.

```bash
read...
network_recv...
cancellable = 1
wait_cancellable...
destroying...
iio_buffer_cancel...
task_stop
task stop: 0 0 id: 28940624
do cancel...
do cancel done
iio_buffer_cancel done
free_buffer...
free1
pfd[1].revents = 0
task stop: 0 1 id: 28626704
```

catkira, 2024-11-05:
something is wrong with the cancel_fd, it has wrong value when poll() is called after cancelling the buffer.

catkira, 2024-11-05:
the other cancel_fd is from the other buffer that my application has open, so thats not the problem. I have to debug a bit more...

catkira, 2024-11-06:
its a bit difficult to debug, because every iio context and every buffer has its own thread which all use functions from network.c and there is no real documentation about the internals of libiio ^^

catkira, 2024-11-06:
@nunojsa I have some more findings, it goes into a 100% cpu load endless loop here https://github.com/analogdevicesinc/libiio/blob/790a6c9324c775e38ab319534d97f0e3253336ff/iiod-responder.c#L571-L572
```c
		for (tmp = priv->readers; tmp->r_next; )
			tmp = tmp->r_next;
```

catkira, 2024-11-06:
in c++ this would just be a std::vector and so much easier to debug

catkira, 2024-11-06:
it looks like the same reader is added twice to the list which causes an endless loop
```bash
[2024-11-06 11:34:23.757] [control_center::iio_buf_reader] [warning] ---     dequeue: timed out    ---
thread = 1778366016, adding reader 179795776
list item 0 = 181190880
list item 1 = 180965088
list item 2 = 179790880
list item 3 = 180897360
list item 4 = 180966064
list item 5 = 180047712
list item 6 = 180763824
list item 7 = 179200736
list item 8 = 180901488
list item 9 = 181200016
list item 10 = 183120608
list item 11 = 178781568
list item 12 = 181129104
list item 13 = 180895728
list item 14 = 179365760
list item 15 = 181301104
thread = 1786758720, wait_cancellable done
b ts: 0 1
c ts: 0 1
x1
thread = 1795151424, poll cancel_fd 20
thread = 1795151424, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 1786758720, network_recv done
thread = 1786758720, network_recv cancellable = 1
thread = 1786758720, wait_cancellable...
thread = 1786758720, poll cancel_fd 20
thread = 1786758720, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1786758720, wait_cancellable done
thread = 1786758720, network_recv done
thread = 1786758720, network_recv cancellable = 1
thread = 1786758720, wait_cancellable...
thread = 1786758720, poll cancel_fd 20
thread = 1786758720, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1786758720, wait_cancellable done
thread = 1786758720, network_recv done
thread = 1786758720, network_recv cancellable = 1
thread = 1786758720, wait_cancellable...
thread = 1786758720, poll cancel_fd 20
thread = 1786758720, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1786758720, wait_cancellable done
x2
task stop: 0 1 id: 31090272
thread = 1786758720, network_recv done
thread = 1786758720, network_recv cancellable = 1
thread = 1786758720, wait_cancellable...
thread = 1786758720, poll cancel_fd 20
thread = 1786758720, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1786758720, wait_cancellable done
thread = 1786758720, network_recv done
thread = 1786758720, read done
thread = 1786758720, read...
thread = 1786758720, network_recv cancellable = 1
thread = 1786758720, wait_cancellable...
thread = 1786758720, poll cancel_fd 20
thread = 1778366016, adding reader 179795776
list item 0 = 181190880
list item 1 = 180965088
list item 2 = 179790880
list item 3 = 180897360
list item 4 = 180966064
list item 5 = 180047712
list item 6 = 180763824
list item 7 = 179200736
list item 8 = 180901488
list item 9 = 181200016
list item 10 = 183120608
list item 11 = 178781568
list item 12 = 181129104
list item 13 = 180895728
list item 14 = 179365760
list item 15 = 181301104
list item 16 = 178759280
thread = 1778366016, discard reader 181190880
[2024-11-06 11:34:23.758] [control_center::iio_buf_reader] [warning] ---     dequeue: timed out    ---
b ts: 0 1
c ts: 0 1
x1
thread = 1795151424, poll cancel_fd 20
thread = 1795151424, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 1778366016, adding reader 181190880
list item 0 = 180965088
list item 1 = 179790880
list item 2 = 180897360
list item 3 = 180966064
list item 4 = 180047712
list item 5 = 180763824
list item 6 = 179200736
list item 7 = 180901488
list item 8 = 181200016
thread = 1795151424, poll cancel_fd 20
thread = 1795151424, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
list item 9 = 183120608
list item 10 = 178781568
list item 11 = 181129104
task stop: 0 1 id: 31090272
list item 12 = 180895728
list item 13 = 179365760
list item 14 = 181301104
list item 15 = 178759280
list item 16 = 179795776
list item 17 = 179795776
list item 18 = 179795776
list item 19 = 179795776
list item 20 = 179795776
list item 21 = 179795776
list item 22 = 179795776
list item 23 = 179795776
list item 24 = 179795776
list item 25 = 179795776
list item 26 = 179795776
list item 27 = 179795776
list item 28 = 179795776
list item 29 = 179795776
list item 30 = 179795776
list item 31 = 179795776
list item 32 = 179795776
list item 33 = 179795776
list item 34 = 179795776
list item 35 = 179795776
list item 36 = 179795776
list item 37 = 179795776
list item 38 = 179795776
list item 39 = 179795776
list item 40 = 179795776
list item 41 = 179795776
list item 42 = 179795776
list item 43 = 179795776
list item 44 = 179795776
list item 45 = 179795776
list item 46 = 179795776
list item 47 = 179795776
list item 48 = 179795776
list item 49 = 179795776
list item 50 = 179795776
list item 51 = 179795776
list item 52 = 179795776
list item 53 = 179795776
list item 54 = 179795776
list item 55 = 179795776
list item 56 = 179795776
list item 57 = 179795776
list item 58 = 179795776
list item 59 = 179795776
list item 60 = 179795776
list item 61 = 179795776
list item 62 = 179795776
list item 63 = 179795776
list item 64 = 179795776
list item 65 = 179795776
list item 66 = 179795776
list item 67 = 179795776
list item 68 = 179795776
list item 69 = 179795776
list item 70 = 179795776
list item 71 = 179795776
list item 72 = 179795776
list item 73 = 179795776
list item 74 = 179795776
list item 75 = 179795776
list item 76 = 179795776
list item 77 = 179795776
list item 78 = 179795776
list item 79 = 179795776
list item 80 = 179795776
list item 81 = 179795776
list item 82 = 179795776
list item 83 = 179795776
list item 84 = 179795776
list item 85 = 179795776
list item 86 = 179795776
list item 87 = 179795776
list item 88 = 179795776
list item 89 = 179795776
list item 90 = 179795776
list item 91 = 179795776
list item 92 = 179795776
list item 93 = 179795776
list item 94 = 179795776
list item 95 = 179795776
list item 96 = 179795776
list item 97 = 179795776
list item 98 = 179795776
list item 99 = 179795776
list item 100 = 179795776
```

```c
	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	if (!priv->readers) {
		priv->readers = io;
	} else {
		unsigned int i = 0;
		for (tmp = priv->readers; tmp->r_next; ) {
			printf("list item %d = %d\n", i++,tmp);
			tmp = tmp->r_next;
			if (i > 100) exit(1);
		}
		tmp->r_next = io;
	}
```

catkira, 2024-11-06:
yes I tried this work around, but then it still hangs sometimes at other locations (probably because of the same underlying reason)
```c
	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	bool found = false;
	if (priv->readers) {
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io)
			{
				printf("reader is already in list!!\n");
				found = true;
			}
			tmp = tmp->r_next;
		}
	}
	if (!found) {
		if (!priv->readers) {
			priv->readers = io;
		} else {
			unsigned int i = 0;
			for (tmp = priv->readers; tmp->r_next; ) {
				printf("list item %d = %d\n", i++,tmp);
				tmp = tmp->r_next;
				if (i > 100) exit(1);
			}
			tmp->r_next = io;
		}
	}
```

catkira, 2024-11-06:
I still get endless loops at the same location even when the same item is not inserted twice
[image]
maybe memory corruption??

catkira, 2024-11-06:
```c
	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	bool found = false;
	if (priv->readers) {
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io)
			{
				printf("reader is already in list!!\n");
				found = true;
				exit(1);
			}
			if (tmp == tmp->r_next) {
				printf("loop detected!!!");
				exit(1);
			}
			tmp = tmp->r_next;
		}
	}
	if (!found) {
		if (!priv->readers) {
			priv->readers = io;
		} else {
			unsigned int i = 0;
			for (tmp = priv->readers; tmp->r_next; ) {
				printf("list item %d = %d\n", i++,tmp);
				tmp = tmp->r_next;
				if (i > 100) exit(1);
			}
			tmp->r_next = io;
		}
	}
```

```bash
thread = 2539071040, adding reader 182425776
loop detected!!!
thread = 2588911168, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2588911168, wait_cancellable done
thread = 2588911168, network_recv done
thread = 2588911168, read done
thread = 2588911168, try lock...
thread = 2588911168, locked
thread = 2588911168, discard reader 182140576
thread = 2588911168, read...
thread = 2588911168, network_recv cancellable = 1
thread = 2588911168, wait_cancellable...
thread = 2588911168, poll cancel_fd 15
thread = 2588911168, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2588911168, wait_cancellable done
thread = 2588911168, network_recv done
thread = 2588911168, read done
thread = 2588911168, read...
thread = 2588911168, network_recv cancellable = 1
thread = 2588911168, wait_cancellable...
thread = 2588911168, poll cancel_fd 15
thread = 3178092864, adding reader 182140576
b ts: 0 1
c ts: 0 1
x1
thread = 2597303872, poll cancel_fd 15
thread = 2597303872, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
task stop: 0 1 id: 32557408
thread = 2588911168, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2588911168, wait_cancellable done
thread = 2588911168, network_recv done
thread = 2588911168, read done
thread = 2588911168, try lock...
thread = 2588911168, locked
thread = 2588911168, discard reader 182140576
thread = 2588911168, read...
thread = 2588911168, network_recv cancellable = 1
thread = 2588911168, wait_cancellable...
thread = 2588911168, poll cancel_fd 15
thread = 2588911168, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2588911168, wait_cancellable done
thread = 2588911168, network_recv done
thread = 2588911168, read done
thread = 2588911168, read...
thread = 2588911168, network_recv cancellable = 1
thread = 2588911168, wait_cancellable...
thread = 2588911168, poll cancel_fd 15
terminate called without an active exception
Aborted (core dumped)
```

catkira, 2024-11-06:
I think io->r_next is set to NULL, because it will always be inserted at the end of the list.

catkira, 2024-11-06:
When I change the code that adds items to the reader list like this, it does not hang anymore.
This `printf("io->r_next = %d\n", io->r_next);` printf showed, that the next item was not set to 0, but instead to itself. Manually setting it to 0 seems to fix it. But why do I have to set it to 0, it is already done before `io->r_next = NULL;` ??
```c
	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	bool found = false;
	if (priv->readers) {
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io)
			{
				printf("reader is already in list!!\n");
				found = true;
				exit(1);
			}
			if (tmp == tmp->r_next) {
				printf("loop detected!!!\n");
				exit(1);
			}
			tmp = tmp->r_next;
		}
	}
	if (!found) {
		if (!priv->readers) {
			priv->readers = io;
		} else {
			unsigned int i = 0;
			for (tmp = priv->readers; tmp->r_next; ) {
				printf("list item %d = %d\n", i++,tmp);
				tmp = tmp->r_next;
				if (i > 100) exit(1);
			}
			tmp->r_next = io;
			printf("io->r_next = %d\n", io->r_next);
			io->r_next = NULL;
		}
	}
```

catkira, 2024-11-06:
you can see my change in the PR, the second to last commit contains all the debug printfs.

pcercuei, 2024-11-06:
If you suspect memory corruption, try to compile with GCC's built-in address sanitizer.
Add `-fsanitize=address` to the C flags, and `-static-libasan` to the linker flags.

If there is any use-after-free, or a buffer overflow, it will detect it.

catkira, 2024-11-06:
I tried it. It freezes without any sanitizer output.

nunojsa, 2024-11-06:
> I think io->r_next is set to NULL, because it will always be inserted at the end of the list.

Sure, it marks the end of the list. My point is that if you're adding an object that already exists on the list and do `io->r_next  = NULL`, you're cutting the list at that point right? So your code that tries to detect that the same object is being added won't really work (unless I'm missing something).

> This printf("io->r_next = %d\n", io->r_next); printf showed, that the next item was not set to 0, but instead to itself.

And this really seems to suggest we're adding an object that is already on the list

pcercuei, 2024-11-06:
> @pcercuei it looks like the same reader is added twice to the list which causes an endless loop

Yeah that's something that is definitely not supported (nor should it be). The real question then, is why is it added twice.

Note that there is one `iiod_responder` instance per buffer created, plus a main `iiod_responder` for things like reading attributes.
When you call `iio_cancel_buffer()`, only the iiod_responder that corresponds to the buffer is canceled. At that point, since the underlying socket is deemed unreliable, the "free block" and "free buffer" commands go through the main `iiod_responder` instead.

I suspect that you are getting a race in the main "iiod_responder", maybe by calling iio_buffer_cancel() while another thread in your program is busy reading an attribute; or by calling iio_buffer_cancel() in a signal handler (that case is explicitely written as unsupported in the doc). From what I can read in the code, iiod_responder does have some multi-thread protection in place, but it does not protect against two threads issueing two requests on the same iiod_io.

catkira, 2024-11-06:
yes I also think that it is some kind of race condition, because I only encounter this when I run the code on a fast CPU.

catkira, 2024-11-06:
> Sure, it marks the end of the list. My point is that if you're adding an object that already exists on the list and do `io->r_next = NULL`, you're cutting the list at that point right? So your code [...]

ah yes, now I understand what you mean, had to read it twice and think a bit :)

catkira, 2024-11-06:
> I suspect that you are getting a race in the main "iiod_responder", maybe by calling iio_buffer_cancel() while another thread in your program is busy reading an attribute; or by calling [...]

I am not calling any iio stuff from signal handlers.

catkira, 2024-11-07:
this code
```c
	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	bool found = false;
	if (priv->readers) {
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io)
			{
				printf("reader is already in list!!\n");
				found = true;
			}
			if (tmp == tmp->r_next) {
				printf("loop detected!!!\n");
			}
			tmp = tmp->r_next;
		}
	}
	if (!found) {
		if (!priv->readers) {
			priv->readers = io;
		} else {
			unsigned int i = 0;
			for (tmp = priv->readers; tmp->r_next; ) {
				printf("list item %d = %d\n", i++,tmp);
				tmp = tmp->r_next;
				if (i > 100) exit(1);
			}
			tmp->r_next = io;
			printf("io->r_next = %d\n", io->r_next);
			// io->r_next = NULL;
		}
	}
	else	exit(1);
```
gives this debug output
```
...
thread = 1553978944, poll cancel_fd 20
thread = 1553978944, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
task stop: 0 1 id: 201282352
thread = 1476392512, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1476392512, wait_cancellable done
thread = 1476392512, network_recv done
thread = 1476392512, network_recv cancellable = 1
thread = 1476392512, wait_cancellable...
thread = 1476392512, poll cancel_fd 20
thread = 1476392512, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1476392512, wait_cancellable done
thread = 1476392512, network_recv done
thread = 1476392512, network_recv cancellable = 1
thread = 1476392512, wait_cancellable...
thread = 1476392512, poll cancel_fd 20
thread = 1476392512, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1476392512, wait_cancellable done
thread = 1476392512, network_recv done
thread = 1476392512, network_recv cancellable = 1
thread = 1476392512, wait_cancellable...
thread = 1476392512, poll cancel_fd 20
thread = 1476392512, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1476392512, wait_cancellable done
thread = 1476392512, network_recv done
thread = 1476392512, read done
thread = 1476392512, read...
thread = 1476392512, network_recv cancellable = 1
thread = 1476392512, wait_cancellable...
thread = 1476392512, poll cancel_fd 20
thread = 1467999808, adding reader 203115920
list item 0 = 202944352
[... 185 lines cut ...]
thread = 1595942464, poll cancel_fd 15
thread = 1595942464, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
task stop: 0 1 id: 52226864
thread = 1587549760, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1587549760, wait_cancellable done
thread = 1587549760, network_recv done
thread = 1587549760, read done
thread = 1587549760, try lock...
thread = 1587549760, locked
thread = 1587549760, discard reader 202002128
thread = 1587549760, read...
thread = 1587549760, network_recv cancellable = 1
thread = 1587549760, wait_cancellable...
thread = 1587549760, poll cancel_fd 15
thread = 1587549760, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1587549760, wait_cancellable done
thread = 1587549760, network_recv done
thread = 1587549760, read done
thread = 1587549760, read...
thread = 1587549760, network_recv cancellable = 1
thread = 1587549760, wait_cancellable...
thread = 1587549760, poll cancel_fd 15
terminate called without an active exception
Aborted (core dumped)
```

catkira, 2024-11-07:
the interesting part is this
```
thread = 1467999808, adding reader 203115920
list item 0 = 202944352
list item 1 = 203407072
list item 2 = 203405440
list item 3 = 203286640
list item 4 = 203392336
list item 5 = 203087776
list item 6 = 203274048
list item 7 = 202356576
list item 8 = 202357072
list item 9 = 202477808
list item 10 = 202478304
list item 11 = 203381504
list item 12 = 203381936
list item 13 = 202940176
list item 14 = 202940672
list item 15 = 203272480
io->r_next = 203115920
thread = 1467999808, discard reader 202944352
loop detected!!! 203115920 -> 203115920
```
203115920 is added to the list and at that moment it looks like it is not already on the list. However when adding it to the list, `io->r_next` is suddenly not NULL anymore !

nunojsa, 2024-11-07:
Could you try this simple diff?

```
diff --git a/iiod-responder.c b/iiod-responder.c
index d654602e67cb..c8840776c6ab 100644
--- a/iiod-responder.c
+++ b/iiod-responder.c
@@ -561,16 +561,24 @@ int iiod_io_get_response_async(struct iiod_io *io,
                memcpy(io->r_io.buf, buf, sizeof(*buf) * nb);
        io->r_io.nb_buf = nb;
        io->r_done = false;
-       io->r_next = NULL;
        io->r_io.start_time = read_counter_us();

        /* Add it to the readers list */
        if (!priv->readers) {
                priv->readers = io;
        } else {
-               for (tmp = priv->readers; tmp->r_next; )
+               for (tmp = priv->readers; tmp->r_next; ) {
+                       if (tmp == io) {
+                               printf("Item(%u) already on the list!\n", io->client_id);
+                               iio_mutex_unlock(priv->lock);
+                               return -EEXIST;
+                       }
+
                        tmp = tmp->r_next;
+               }
+
                tmp->r_next = io;
+               io->r_next = NULL;
        }

        iio_mutex_unlock(priv->lock);
```

catkira, 2024-11-07:
this just works, exit(1) never gets called.
```c
	io->r_io.nb_buf = nb;
	io->r_done = false;
	// io->r_next = NULL;
	io->r_io.start_time = read_counter_us();

	/* Add it to the readers list */
	printf("thread = %u, adding reader %d\n", pthread_self(), io);

	if (!priv->readers) {
		priv->readers = io;
	} else {
		unsigned int i = 0;
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io) {
					printf("Item(%u) already on the list!\n", io->client_id);
					iio_mutex_unlock(priv->lock);
					exit(1);
					return -EEXIST;
			}
			tmp = tmp->r_next;
		}
		tmp->r_next = io;
		io->r_next = NULL;
	}

	iio_mutex_unlock(priv->lock);
```

catkira, 2024-11-07:
Why does it make a difference whether we set the r_next pointer to NULL before or after the element is added to the list? I understand that it could make a difference if the element is added twice to the list, but this is according to my debug output not the case.

catkira, 2024-11-07:
another run with this code:
```c
	printf("thread = %u, adding reader %d\n", pthread_self(), io);
	bool found = false;
	if (priv->readers) {
		for (tmp = priv->readers; tmp->r_next; ) {
			if (tmp == io)
			{
				printf("reader is already in list!!\n");
				found = true;
			}
			if (tmp == tmp->r_next) {
				printf("loop detected!!!\n");
			}
			tmp = tmp->r_next;
		}
	}
	if (!found) {
		if (!priv->readers) {
			priv->readers = io;
		} else {
			unsigned int i = 0;
			for (tmp = priv->readers; tmp->r_next; ) {
				printf("list item %d = %d\n", i++,tmp);
				tmp = tmp->r_next;
				if (i > 100) exit(1);
			}
			tmp->r_next = io;
			printf("io->r_next = %d\n", io->r_next);
			// io->r_next = NULL;
		}
	}
	else	exit(1);
```

```
thread = 1409283648, discard reader 183644016
thread = 1409283648, read...
thread = 1409283648, network_recv cancellable = 1
thread = 1409283648, wait_cancellable...
thread = 1409283648, poll cancel_fd 20
thread = 1400890944, adding reader 183428592
list item 0 = 183643312
list item 1 = 183429264
list item 2 = 183429760
list item 3 = 183769424
list item 4 = 183769920
list item 5 = 183770416
list item 6 = 183501328
list item 7 = 183501744
list item 8 = 183502240
list item 9 = 183502736
list item 10 = 183247120
list item 11 = 183247616
list item 12 = 183248112
list item 13 = 183248608
io->r_next = 0
thread = 1400890944, discard reader 183644016
b ts: 0 1
c ts: 0 1
x1
thread = 1486870080, poll cancel_fd 20
thread = 1486870080, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 1486870080, poll cancel_fd 20
thread = 1486870080, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
task stop: 0 1 id: 181671728
thread = 1400890944, adding reader 183644016
list item 0 = 183643312
list item 1 = 183429264
[... 96 lines cut ...]
list item 12 = 183248112
list item 13 = 183248608
list item 14 = 183249104
list item 15 = 183428592
io->r_next = 183644016
b ts: 0 1
c ts: 0 1
x1
thread = 1486870080, poll cancel_fd 20
thread = 1486870080, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 1486870080, poll cancel_fd 20
thread = 1486870080, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
task stop: 0 1 id: 181671728
thread = 1400890944, discard reader 183643312
loop detected!!! 183644016 -> 183644016
QObject::~QObject: Timers cannot be stopped from another thread
thread = 1409283648, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 1409283648, wait_cancellable done
thread = 1409283648, network_recv done
thread = 1409283648, read done
thread = 1409283648, try lock...
terminate called without an active exception
Aborted (core dumped)
```

catkira, 2024-11-07:
it looks like "183644016" is really added twice, but why does it not show up in my list dump no the 2nd add???

nunojsa, 2024-11-07:
Hmm, sou you never see the printf()?! 

> I understand that it could make a difference if the element is added twice to the list, but this is according to my debug output not the case.

Exactly, it would matter if the element is added twice. IMO; it's still a good check to have rather thann corrupting the list and adding an infinite loop.

> In my understanding the only reason could be, that another thread modifies the io item, after r_next is set to NULL and before tmp->r_next = io;. This is a very short period. Therefore it probably only happens on fast CPUs.

Well, that should not logically be possible (I guess) unless we have a bug somewhere and change the object without the proper `iio_mutex_lock(priv->lock);`.

catkira, 2024-11-07:
> Hmm, sou you never see the printf()?!

never

catkira, 2024-11-07:
Ah my way of recursing the linked list is wrong. It has to be `for (tmp = priv->readers; tmp; ) {` and not `for (tmp = priv->readers; tmp->r_next; ) {`

catkira, 2024-11-07:
now I get this output
```
thread = 4026529344, poll cancel_fd 20
thread = 4018136640, adding reader 193412560
list item 0 = 193595232
list item 1 = 192262704
list item 2 = 191784992
list item 3 = 192263456
list item 4 = 193750496
list item 5 = 193671392
list item 6 = 192860624
list item 7 = 193402800
list item 8 = 193399264
list item 9 = 193399760
list item 10 = 192859040
list item 11 = 192859472
list item 12 = 193740224
list item 13 = 193740672
list item 14 = 192853056
io->r_next = 0
thread = 4018136640, discard reader 193595232
thread = 4018136640, adding reader 193595232
list item 0 = 192262704
list item 1 = 191784992
list item 2 = 192263456
list item 3 = 193750496
list item 4 = 193671392
list item 5 = 192860624
list item 6 = 193402800
list item 7 = 193399264
list item 8 = 193399760
list item 9 = 192859040
list item 10 = 192859472
list item 11 = 193740224
list item 12 = 193740672
list item 13 = 192853056
list item 14 = 195898464
io->r_next = 0
b ts: 0 1
c ts: 0 1
x1
thread = 4102399552, poll cancel_fd 20
thread = 4102399552, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 4102399552, poll cancel_fd 20
thread = 4102399552, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
x1
thread = 4102399552, poll cancel_fd 20
thread = 4102399552, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
task stop: 0 1 id: 42785376
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
thread = 4026529344, network_recv done
thread = 4026529344, read done
thread = 4026529344, try lock...
thread = 4026529344, locked
thread = 4026529344, discard reader 195898464
thread = 4026529344, read...
thread = 4026529344, network_recv cancellable = 1
thread = 4026529344, wait_cancellable...
thread = 4026529344, poll cancel_fd 20
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
thread = 4026529344, network_recv done
thread = 4026529344, network_recv cancellable = 1
thread = 4026529344, wait_cancellable...
thread = 4026529344, poll cancel_fd 20
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
[... 342 lines cut ...]
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
thread = 4026529344, network_recv done
thread = 4026529344, network_recv cancellable = 1
thread = 4026529344, wait_cancellable...
thread = 4026529344, poll cancel_fd 20
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
thread = 4026529344, network_recv done
thread = 4026529344, read done
thread = 4026529344, read...
thread = 4026529344, network_recv cancellable = 1
thread = 4026529344, wait_cancellable...
thread = 4026529344, poll cancel_fd 20
thread = 4018136640, adding reader 192262704
reader is already in list!!
QObject::~QObject: Timers cannot be stopped from another thread
thread = 4026529344, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 4026529344, wait_cancellable done
thread = 4026529344, network_recv done
thread = 4026529344, read done
thread = 4026529344, try lock...
terminate called without an active exception
Aborted (core dumped)
```

catkira, 2024-11-07:
this part is also interesting, the same reader is removed twice, from separate threads:
```
thread = 4018136640, adding reader 193595232
list item 0 = 192262704
list item 1 = 191784992
list item 2 = 192263456
list item 3 = 193750496
list item 4 = 193671392
list item 5 = 192860624
list item 6 = 193402800
list item 7 = 193399264
list item 8 = 193399760
list item 9 = 192859040
list item 10 = 192859472
list item 11 = 193740224
list item 12 = 193740672
io->r_next = 0
b ts: 0 1
c ts: 0 1
x1
thread = 4026529344, locked
thread = 4026529344, discard reader 192262704
thread = 4026529344, read...
thread = 4026529344, network_recv cancellable = 1
thread = 4026529344, wait_cancellable...
thread = 4026529344, poll cancel_fd 20
thread = 4102399552, poll cancel_fd 20
thread = 4102399552, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
thread = 4018136640, discard reader 192262704
thread = 4102399552, poll cancel_fd 20
thread = 4102399552, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
x2
```

nunojsa, 2024-11-07:
> Ah my way of recursing the linked list is wrong. It has to be for (tmp = priv->readers; tmp; ) { and not for (tmp = priv->readers; tmp->r_next; ) {, therefore the last element was always missing in my list dump.

Oh yeah! That explains the whole story!

catkira, 2024-11-07:
and then later, the same reader is added twice. So the problem already starts when two different threads try to remove the reader from the list, they probably both think they own the io item.

catkira, 2024-11-07:
I think this block needs to be an exclusive section
https://github.com/analogdevicesinc/libiio/blob/790a6c9324c775e38ab319534d97f0e3253336ff/iiod-responder.c#L287-L305

catkira, 2024-11-07:
ok, I am pretty close to the problem. I added another debug printf here
```c
static ssize_t iiod_client_read_attr_new(struct iiod_client *client,
					 const struct iio_attr *attr,
					 char *dest, size_t len)
{
	struct iiod_io *io = iiod_responder_get_default_io(client->responder);
	printf("* thread = %u, iiod_responder_get_default_io = %d\n", pthread_self(), io);
```

with this I get
```
thread = 2743068224, wait_cancellable...
thread = 2743068224, poll cancel_fd 15
* thread = 3404880192, iiod_responder_get_default_io = 185775904thread = 3404880192, adding reader 185775904
thread = 2751460928, poll cancel_fd 15
thread = 2751460928, pfd[0].revents = 4, pfd[1].revents = 0, ret = 1
task stop: 0 1 id: 184277408
thread = 2743068224, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2743068224, wait_cancellable done
thread = 2743068224, network_recv done
thread = 2743068224, read done
thread = 2743068224, try lock...
thread = 2743068224, locked
* thread = 2743068224, selecting io 185775904 with client_id 0
thread = 2743068224, discard reader 185775904
thread = 2743068224, read...
thread = 2743068224, network_recv cancellable = 1
thread = 2743068224, wait_cancellable...
thread = 2743068224, poll cancel_fd 15
thread = 2743068224, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2743068224, wait_cancellable done
thread = 2743068224, network_recv done
thread = 2743068224, read done
thread = 2743068224, read...
thread = 2743068224, network_recv cancellable = 1
thread = 2743068224, wait_cancellable...
thread = 2743068224, poll cancel_fd 15
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, read done
thread = 2701104704, try lock...
thread = 2701104704, locked
* thread = 2701104704, selecting io 186528768 with client_id 13
thread = 2701104704, discard reader 186528768
thread = 2701104704, read...
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
[... 225 lines cut ...]
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, read done
thread = 2701104704, read...
thread = 2701104704, network_recv cancellable = 1
thread = 2701104704, wait_cancellable...
thread = 2701104704, poll cancel_fd 20
thread = 2617243200, adding reader 185559840
reader is already in list!!
QObject::~QObject: Timers cannot be stopped from another thread
thread = 2701104704, pfd[0].revents = 1, pfd[1].revents = 0, ret = 1
thread = 2701104704, wait_cancellable done
thread = 2701104704, network_recv done
thread = 2701104704, read done
thread = 2701104704, try lock...
terminate called without an active exception
Aborted (core dumped)
```
there seems to be a race condition between two threads for an io element.

catkira, 2024-11-07:
there needs to be one location (a pool or factory) where io elements are taken from and this location needs to be exclusive (mutexed).

catkira, 2024-11-07:
@nunojsa @pcercuei r u going to rework this or should I do it? (I would prefer if u do it, because I spent a lot of time on this already)

pcercuei, 2024-11-07:
Somebody at ADI can fix it :)

nunojsa, 2024-11-07:
I think there might be a race with `io->r_done`. In `iiod_io_get_response_async()`, we only use `priv->lock` which seems to be the one used for the linked list of io elements. But on all (I think?!) other accesses to `io->r_done`, it seems we need  `iio_mutex_lock(io->lock);`. For instance look at `iiod_responder_signal_io()`. That can race with `iiod_io_get_response_async()`. Unless, `iiod_responder_signal_io()` is always called with priv->lock held but that would still be questionable.

> @nunojsa @pcercuei r u going to rework this or should I do it? (I would prefer if u do it, because I spent a lot of time on this already)

Paul is no longer in ADI so I dunno... I can take a look but only in the next few weeks

catkira, 2024-11-07:
> I can take a look but only in the next few weeks

:'(

pcercuei, 2024-11-07:
> I think there might be a race with `io->r_done`. In `iiod_io_get_response_async()`, we only use `priv->lock` which seems to be the one used for the linked list of io elements. But on all (I think?!) [...]

The `io->lock` is only used alongside the `io->cond` as a notification mechanism when the response has been obtained. In `iiod_io_get_response_async()` you're not yet waiting for any response so it doesn't have to be locked.

I think you'd need an extra mutex to protect an "iiod_io" from being used twice. E.g. lock it in `iiod_io_get_response_async()`, unlock it in `iiod_io_wait_for_response()` and `iiod_responder_signal_io()`.

catkira, 2024-11-07:
I implemented an additional mutex and it seems to work.

catkira, 2024-11-07:
I tried it quickly, but it is not so easy. object oriented programming with c is a pain. self implemented containers with c are a pain. multithreading with c is a pain.

catkira, 2024-11-07:
My current attempt can be seen here: https://github.com/analogdevicesinc/libiio/pull/1212
it does not crash anymore, but it sometimes hangs due to mutex lock ^^

nunojsa, 2024-11-07:
To me, we already have the readers list so we should likely centralize adding and removing io's to that list and those places would be the ultimate lock()/unlock() points. Might need some refactors in other places but logically it makes sense to me.

> it does not crash anymore, but it sometimes hangs due to mutex lock ^^

Yeah, I was wondering the impact of that serialization with the new lock... I think the main problem might be the default_io because, well, being the default, we might have more threads using it. So, we could think about having an io pool of "defaults" (let's call it like that :)) properly serialized and get different io's in `iiod_command_get_default_io()` (properly renamed). We would then also need to be careful (and understand) when we can give back the io to the pool. Is the hanging extremely annoying :sweat_smile: ?

That said, not sure about more subtle implications of having a pool rather than a single default_io

pcercuei, 2024-11-07:
> To me, we already have the readers list so we should likely centralize adding and removing io's to that list and those places would be the ultimate lock()/unlock() points. Might need some refactors in other places but logically it makes sense to me.

But the problem is actually when different threads enqueue a response request using the same iiod_io. The `priv->lock` should already protect the insertion/removal from the readers list.

> Yeah, I was wondering the impact of that serialization with the new lock... I think the main problem might be the default_io because, well, being the default, we might have more threads using it. [...]

You could have one iiod_io dedicated for the block/buffer removal.

catkira, 2024-11-07:
It does not work with a simple mutex. A simple mutex can only prevent double insertions into the list, but then there are still occasional mutex deadlocks.

`iiod_command_get_default_io()` in responder.c is called by multiple threads. What prevents multiple threads using the same io?

catkira, 2024-11-07:
It does not crash (without any additional mutex) when I create a new io item with every call to iiod_responder_get_default_io()
```c
struct iiod_io *
iiod_responder_get_default_io(struct iiod_responder *priv)
{
	struct iiod_io* io = iiod_responder_create_io(priv, 0);
	io->timeout_ms = priv->timeout_ms;
	return io;
}
```
but thats also inefficient and a memory leak, because they never get deleted. Doing libiio in c++ would be 100x easier to maintain, because one could use standard containers and smart/unique pointers.

catkira, 2024-11-07:
it is never really clear who owns a pointer with this c-style. If I would create an io pool in iiod_responder, then how do the items get returned to the pool?? All the functions that use an io object would need to keep track of ownership, so that the last function that uses it can return it to the pool. Thats so easy with smart/unique pointers but a complete mess with just raw pointers. UNMAINTANABLE!

nunojsa, 2024-11-07:
> But the problem is actually when different threads enqueue a response request using the same iiod_io. The priv->lock should already protect the insertion/removal from the readers list.

I know, I was talking about the sync lock you spoke about. I'm assuming that as long as an iiod_io is on readers list, it is a busy one and cannot be re-used.

> It does not crash (without any additional mutex) when I create a new io item with every call to iiod_responder_get_default_io()

Yeah, that's why I was speaking about a pool of iiod_io. 

> struct iiod_io* io = iiod_responder_create_io(priv, 0);

Hmm, not sure if we can hardcode the client_id always to 0... 

>  Doing libiio in c++ would be 100x easier to maintain, because one could use standard containers and smart/unique pointers.

Well, arguable :smiley:

catkira, 2024-11-07:
and implementing shared pointer behaviour with manual ref counting is a total mess too !!!

catkira, 2024-11-07:
who can maintain these scattered `iiod_io_ref_unlocked()` and  `iiod_io_unref_unlocked()`, I am getting a meltdown ^^

[36 more replies]