analogdevicesinc/libiio #1215: Permissions issue libiio python

ianzur, 2024-11-22:
I am unable to create a buffer with pylibiio unless I run my code with elevated permissions.

Raspberry Bookworm 6.6.62, installed via apt (python3-libiio)
iio backend version: v24.4

<details>
<summary> Testing with python example code </summary>

To demonstrate I pulled the example python code in this repo. I am also unable to create a buffer in my attempts. 

Example code: https://github.com/analogdevicesinc/libiio/tree/main/bindings/python/examples

More info regarding my situation, MPU9250 (IMU) is connect via i2c and using the inv-mpu6050 driver
```sh
$ python3 iio_info.py
IIO context created: local
Backend version: 0.24 (git tag: v0.24
Backend description string: Linux raspberrypi 6.6.62-v8+ #2 SMP PREEMPT Wed Nov 20 21:29:34 CST 2024 aarch64
IIO context has 2 attributes:
	local,kernel: 6.6.62-v8+
	uri: local:
IIO context has 2 devices:
	iio:device0: mpu9250
		12 channels found:
			accel_x:  (input)
			6 channel-specific attributes found:
				calibbias, value: -7496
				matrix, value: 0, 0, 0; 0, 0, 0; 0, 0, 0
				mount_matrix, value: 1, 0, 0; 0, 1, 0; 0, 0, 1
				raw, value: 5360
				scale, value: 0.000598
				scale_available, value: 0.000598 0.001196 0.002392 0.004785
...
```

This shouldn't error:
```sh
$ python3 iio_readdev.py -b 1 -T 100 mpu9250 0
Traceback (most recent call last):
  File "/home/pi/iio_readdev.py", line 254, in <module>
    main()
  File "/home/pi/iio_readdev.py", line 248, in main
    reader = DataReader(context_builder.create(), arguments)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/iio_readdev.py", line 219, in __init__
    self.buffer = buffer_builder.create()
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/iio_readdev.py", line 196, in create
    buffer = iio.Buffer(self.dev, self.arguments.buffer_size)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/iio.py", line 981, in __init__
    self._buffer = _create_buffer(device._device, samples_count, cyclic)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/iio.py", line 56, in _check_null
    raise OSError(err, _strerror(err))
PermissionError: [Errno 13] Permission denied
```

Elevating permission works:
```sh
$ sudo python3 iio_readdev.py -b 1 -T 100 mpu9250 0
WARNING: High-speed mode not enabled
&@J|tV5

;'JWumWD/12FOa)"GAWPOcPG$*;<@P_WT @,G_J<<B2V[J-QD;?AP5B6J;6=PFK9C>?IU=F_69VL=K/53]]GL8MQKQ615CS\@i6-FI4L=G'?HUSn8NEEH=1QJU?N.JOQgCB/DX>7NJHGQ9\g/TD8[_L`IGPZ[`R9CEVfNG:F>:;MH8_OWZ:6IH5X4MG908KE>P4;BV98>nZUC,+IW_FW:=)6L=PX;/XRH]POH;@eIdTfJ==XXFGQ@6_J4G@NbkT.<16hhTy9U3dGQGf=[I- 4QDNZ-6910?>QPZ8R@;7LAE?^?_PC5YX:G=a\OUIf<e\9>?JDO9EG,D@BBA@PF>DI?IXE1E>*VGQHM:MMCJIZ3>-
...
```

</details>

I receive the same permission error when using the c/c++ equivalents of the above example (iio_readev) packaged in libiio-utils. Again elevating permissions solves the issue.

ianzur, 2024-11-26:
Creating an iio group and adding the user along with the addition of some udev rules solves my permissions issues. 
```sh
# groupadd iio
# usermod -aG iio $USER
```

/etc/udev/rules.d/90-iio.rules
```
# copy owner permissions to group (chmod g=u ...) change group to iio (chgrp iio)
SUBSYSTEM=="iio", RUN+="/bin/sh -c 'chgrp -R iio /sys/bus/iio/devices/$kernel/ && chmod -R g=u /sys/bus/iio/devices/$kernel'"
SUBSYSTEM=="iio", KERNEL=="iio:device*", RUN+="/bin/sh -c 'chgrp iio /dev/$kernel && chmod g=u /dev/$kernel'"
```

Are there security concerns around modifying the group and permissions of the files? As a userspace library to access iio devices, shouldn't these rules, or some equivalent, be shipped with these packages?

rgetz, 2024-11-27:
It depends on your package/distribution maintainer - if they decided not to include them - you need to check with them why.

the rules should be installed on the default images that are built as part of the project on github - if they aren't - you are right - it's a bug.

closed 2025-02-24 by dNechita