analogdevicesinc/libiio #1217: File not found when connecting to device

HassanSlim, 2024-12-02:
Hi,
I am building a yocto kirkstone image on xilinx zynq 7020 device. 
I am facing the following error when trying to read from a register.

````
File "util/fpga_util.py", line 16, in read_modify_write
    old_value = reg_access.reg_read(addr)
  File "util/iio.py", line 1181, in reg_read
    _d_reg_read(self._device, reg, _byref(value))
  File "util/iio.py", line 62, in _check_negative
    raise OSError(-result, _strerror(-result))
FileNotFoundError: [Errno 2] No such file or directory
````

The IIO versions that I tried till now are:
  - v0.21 (since it running fine on zeus build)
  - v0.23 (since it is the version included in oe for kirkstone)
The attached error is coming from version 0.23.

The python version which is used in zeus is 3.7, while in kirkstone is v3.10 (if this plays a role)
Running iio_info in kirkstone I get the following:

````
$ iio_info
Library version: 0.23 (git tag: 92d6a35)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.23 (git tag: 92d6a35)
Backend description string: Linux AIM3020-A01-0001 5.15.36-xilinx-v2022.2 #1 SMP PREEMPT Mon Oct 3 07:50:07 UTC 2022 armv7l
IIO context has 2 attributes:
        local,kernel: 5.15.36-xilinx-v2022.2
        uri: local:
IIO context has 3 devices:
        iio:device0: xadc
                9 channels found:
                        .... (I removed them because not related to my question)
        iio:device1: datacap (buffer capable)
                        .... (I removed them because not related to my question)
        iio:device2: streaming_core (buffer capable)
                1 channels found:
                        voltage0: position_stream (input, index: 0, format: le:U32/32>>0)
                        1 channel-specific attributes found:
                                attr  0: label value: position_stream
                2 device-specific attributes found:
                                attr  0: data_lost value: 0
                                attr  1: enable_queue ERROR: Permission denied (13)
                3 buffer-specific attributes found:
                                attr  0: data_available value: 4096
                                attr  1: length_align_bytes value: 8
                                attr  2: watermark value: 512
                No trigger on this device
```

The device that I am trying to access is the `streaming_core` .
Any help is appreciated.

Best regards,
Hassan

HassanSlim, 2024-12-02:
```
$ ls /sys/kernel/debug/iio/iio:device2/
ls: /sys/kernel/debug/iio/iio:device2/: No such file or directory
```

I didn't mount debugfs, I didn't expect that debugfs is required for normal operation of IIO

mhennerich, 2024-12-02:
Reading and writing registers is not a normal operation.
Therefore it's part of debugfs, and needs to be specifically mounted by root. 

https://elixir.bootlin.com/linux/v6.13-rc1/source/Documentation/ABI/testing/debugfs-iio-backend

HassanSlim, 2024-12-04:
Thank you for your support. I activated the debugfs and problem is solved.

closed 2024-12-04 by HassanSlim