analogdevicesinc/libiio #1218: Is the Libiio support ADRV9025?

deltaRing, 2024-12-07:
Hello community,
 I am using the zcu102 + adrv9026 and libiio for capturing and transmitting signal. However, i can emit nothing but noise during using the libiio. The code is modified based on the adrv9009-iostream.c. Now my question is the libiio c code support adrv9026? Any suggestion is appreciated!

deltaRing, 2024-12-08:
can anyone help me😭😭😭？

deltaRing, 2024-12-08:
okay, looks like my address of the txbuf is changed in each iteration, i'm try to find out what happening to txbuf

ekigwana, 2024-12-08:
> okay, looks like my address of the txbuf is changed in each iteration, i'm try to find out what happening to txbuf

Search through the support forums here https://ez.analog.com/rf/wide-band-rf-transceivers/design-support-adrv9026

And be sure to search for similar "issues" for the AD9361 and you will find that several before you have encountered this and usually it comes down to how the hardware is configured and understanding that. The Engineer Zone is the best place for best effort support.

deltaRing, 2024-12-09:
okay, guys, i've figured out what's the problem of my adrv9025. 

First, adrv9025 utilized the LSB data only, so the above examples utilize MSB would not work correctly.
[image]

Then, the adrv9025 TX4 utilize LO2 by default, so i've set both LO1 and LO2 to the same frequency（That's why i can't receive any data due to i've set LO1 only）
[image]

Next, i've noticed the hardwaregain is low, and i increased it to 0.0dB
[image]

finally, i can receive the signal in my another device (ad9361)
[image]
[image]

As for the above address problem, i've tested it, and it can be my mistake. I input wrong address. everything is pretty normal now.

Thank you for your reading and helping!

closed 2024-12-09 by deltaRing