analogdevicesinc/libiio #1219: macOS 14 libiio support

alikarb, 2024-12-07:
Hi, is libiio supported on macOS 14? I'm trying to get libiio installed so I can talk to PLUTO sdr through python. I'm getting errors about symbol not found. I think it is related to libiio installation

tfcollins, 2024-12-08:
Existing Pkg's should work on 14. If you have an issue please post the full error and versions of library used

alikarb, 2024-12-08:
which pkg file should I be downloading? I'm referencing the release page in the git repo linked below but dont see macos 14 listed? I've also tried to build and install libiio from the source but regardless of what I do, I get an error 
that libiio package is not found. I've attached an odt file with my terminal output

https://github.com/analogdevicesinc/libiio/releases/tag/v0.26

[libiio_install.odt](https://github.om/user-attachments/files/18050851/libiio_install.odt)

tfcollins, 2024-12-10:
Try the macos13 for your systems architecture. Your attached file or link to it is broken

alikarb, 2024-12-11:
Here is the attached file again from the original response.
[libiio.txt](https://github.com/user-attachments/files/18097789/libiio.txt)

I've tried installing macos13 arm pkg file (chip on macbook is m3 pro which I believe is ARM64). I dont think this worked and the reason being is once the installation is complete, I opened a new terminal session and typed 'iio_info' to see if anything shows up. I get the same error that the command is not found. Exact error message is "zsh: command not found: iio_info".

Do I need to add the install path to ~/.zshrc? if so, how can I find the install path?

tfcollins, 2024-12-11:
CLI Tools are available in /Library/Frameworks/iio.framework/Tools . That needs to be on your path or you need to run the tools from their.

For your build disable the python bindings and restart. I would also assume you want libiio v1 based on your build and have read the README.

alikarb, 2024-12-12:
I dont have a Frameworks folder under /Library. At what point should that show up?

I'm not attached to a specific version. My end goal is to be able to communicate with my ADALM Pluto via python (specifically python 3.12)

alikarb, 2024-12-14:
so fortunately my macbook is relatively new and I did not have much installed on here. I decided to do a factory reset and start this install from scratch by directly downloading the .pkg instead of building from source files. I am now able to see devices using CLI tools and communicate with them via python library. Thanks for the help!

closed 2024-12-14 by alikarb