analogdevicesinc/libiio #1224: WARNING: High-speed mode not enabled - kirkstone

HassanSlim, 2025-01-09:
Hi,
I am building a yocto kirkstone image on xilinx zynq 7020 device.
I am facing the following warning:

`WARNING: High-speed mode not enabled`

This warning is causing later a timeout in python

` Timeout hit during buffer.refill()`

I don't get such a warning and timeout error on zeus firmware, but only with kirkstone firmware.
I noticed that if I increased the timeout so much I don't get the error. This means that refilling the buffer is extremly slow.

The IIO versions that I tried till now are:

v0.21 (since it running fine on zeus build but getting the mentioned error above)
v0.23 (since it is the version included in oe for kirkstone)
The attached error is coming from version 0.21.

Did I miss any special configuration while building kirkstone, or is this any setting for IIO.
Thank you in advance for your help.

Best regards,
Hassan

dNechita, 2025-01-10:
Hi Hassan,

Are running your Python script directly on a zynq 7020 device? Or are you running it on a PC that is connected to the device via USB or network?

Can you please share more details about the data payload that you are trying to capture? For example: the sample rate, size (in bytes) of the sample and the number of samples that the buffer is configured with?

What is the timeout value for which you are no longer getting timeouts?

Thanks,
Dan

dNechita, 2025-03-13:
Closing this due to no response. Please feel free to reopen this if you need to.

closed 2025-03-13 by dNechita