analogdevicesinc/libiio #1236: Installers on README being used erroneously

tfcollins, 2025-01-28:
We keep getting users that download development releases from the main readme instead of the releases page. This probably wasn't an issue before the v1 development but has created a lot of issues recently. (Primarily since the libiio python bindings are broken on Windows). I would argue that we put a bigger banner that is "DEVELOPMENT BUILDS ONLY" or use a separate page for them since the are really shouldn't be used by most people.

Comments?

rgetz, 2025-01-29:
That's why the [Readme](https://github.com/analogdevicesinc/libiio/blob/main/README.md) clearly states:

```
If you want to just use libiio, we suggest using the [latest release](https://github.com/analogdevicesinc/libiio/releases/latest).
```

Can't help people who don't read...

-Robin

tfcollins, 2025-01-29:
> That's why the [Readme](https://github.com/analogdevicesinc/libiio/blob/main/README.md) clearly states:
> 
> ```
> If you want to just use libiio, we suggest using the [latest release](https://github.com/analogdevicesinc/libiio/releases/latest).
> ```
> 
> Can't help people who don't read...
> 
> -Robin

I explain that but it goes over like a lead balloon...

rgetz, 2025-01-30:
My suggestion would be to take the entire table out of the readme.md file and the put it into a developer.md file then.

dNechita, 2025-02-04:
I have moved the entire table to another readme file.
I’d like to add a new table to the main README that highlights a few of the latest releases along with a brief list of supported operating systems for each. Since the README serves as a landing page for many users, this would make the latest release more noticeable and encourage them to click on it. The table looks like this: https://github.com/analogdevicesinc/libiio/blob/move-readme-installers-table/README.md

Please see here: https://github.com/analogdevicesinc/libiio/compare/main...move-readme-installers-table

rgetz, 2025-02-06:
I think that's a great start 
 
- I would make the main readme have links to the downloads.
- I would remove the "Unsupported. Last artifacts from xxx" in the developer readme - if it's not being built - it's not interesting to a developer. (IMHO), people can go back to the earlier release to get things.

closed 2025-02-24 by dNechita