analogdevicesinc/libiio #1251: .github/workflows/freebsd.yml not being run?

rgetz, 2025-03-01:
It doesn't look like we are testing CI on freeBSD anymore?

Thanks for checking

tfcollins, 2025-03-02:
It's running in main but not the v0 based stuff.

rgetz, 2025-03-03:
Where do I see the results - I don't see them on the PRs to main?

tfcollins, 2025-03-03:
Look like they only apply to pushes https://github.com/analogdevicesinc/libiio/blob/main/.github/workflows/freebsd.yml#L4 

I'm guessing you are looking at a PR from a fork and not seeing it?

rgetz, 2025-03-04:
yeah, was looking at all the PRs - I see it now on ADI people who make PRs:
https://github.com/analogdevicesinc/libiio/pull/1249/checks

but not on others (from external forks):
https://github.com/analogdevicesinc/libiio/pull/1250/checks

or from PRs labeled as "Draft":
https://github.com/analogdevicesinc/libiio/pull/1246/checks

Is it only done on push on purpose?

I would think it should be more similar to the [others](https://github.com/analogdevicesinc/libiio/blob/main/azure-pipelines.yml)?

tfcollins, 2025-03-05:
I doubt it. We can add it to PRs.

However, they will require a maintainer or collaborator approval to run when PRs are external.

closed 2025-03-11 by dNechita