analogdevicesinc/libiio #1253: libiio on Raspberry Pi

nguyben, 2025-03-06:
I am attempting to use an SDR similar to the PLUTO on a raspberry pi and have been unable to install libiio. Is it even possible to do so and if so, is there a guide?

tfcollins, 2025-03-06:
Yes. Just use the standard build guide https://github.com/analogdevicesinc/libiio/blob/libiio-v0/README_BUILD.md

rgetz, 2025-03-08:
or the existing binary build/release:

- [libiio-0.26.g-Ubuntu-arm32v7.deb](https://github.com/analogdevicesinc/libiio/releases/download/v0.26/libiio-0.26.g-Ubuntu-arm32v7.deb)
- [libiio-0.26.g-Ubuntu-arm32v7.tar.gz](https://github.com/analogdevicesinc/libiio/releases/download/v0.26/libiio-0.26.g-Ubuntu-arm32v7.tar.gz)
- [libiio-0.26.g-Ubuntu-arm64v8.deb](https://github.com/analogdevicesinc/libiio/releases/download/v0.26/libiio-0.26.g-Ubuntu-arm64v8.deb)
- [libiio-0.26.g-Ubuntu-arm64v8.tar.gz](https://github.com/analogdevicesinc/libiio/releases/download/v0.26/libiio-0.26.g-Ubuntu-arm64v8.tar.gz)

dNechita, 2025-03-17:
Closing this. Please feel free to reopen if you are still facing issues.

closed 2025-03-17 by dNechita