analogdevicesinc/libiio #1259: local backend with cma

xR3b0rn, 2025-03-19:
In the current version of **libiio**, the local backend allocates DMA buffers using [[/dev/dma_heap/system](https://github.com/analogdevicesinc/libiio/blob/9f24ce5700bc9f13959a830efd10723a427875de/local-dmabuf.c#L87)](https://github.com/analogdevicesinc/libiio/blob/9f24ce5700bc9f13959a830efd10723a427875de/local-dmabuf.c#L87), which results in a scattered DMA buffer using `sg_table`. However, some hardware requires **contiguous** DMA buffers. In such cases, it would be preferable to use `/dev/dma_heap/cma,linux` instead. Therefore, there should be a way to configure `libiio` to use CMA-based DMA buffers when needed.

dNechita, 2025-03-19:
Hi,
Would it be sufficient if this configuration were available at the time libiio is being built?

xR3b0rn, 2025-03-19:
For my current needs, however, it would be helpful to have a quick upstream solution where a simple CMake option would be sufficient.

dNechita, 2025-03-20:
Hi @xR3b0rn,

I made a draft with a possible implementation: https://github.com/analogdevicesinc/libiio/pull/1260
Can you please have a look and let me know if this works for you?
Thanks!