analogdevicesinc/libiio #1261: Guidance on Reading Continuous 1-Second Data with libiio at 61.44M Sample Rate

whale1st, 2025-03-20:
I'm working on a project where I need to read continuous 1-second data using libiio with the following specifications:

Sampling rate: 61.44 MHz
Bandwidth: 56 MHz
Duration: 1 second

I suspect my current implementation may be losing samples, particularly in the read function where the memcpy operation might be taking too long. I'm unsure if my approach is correct and would appreciate guidance on the proper way to implement this.

Current Implementation:

The current total execution time of the device_receive_samples function is 1.4s.

Below is the relevant part of my code where I handle the reading of data:

```c
config.center_freq = 2400000000ULL;  // 2.4GHz
config.sampling_rate = 61440000;     // 61.44MHz
config.bandwidth = 56000000;         // 56MHz
config.gain_mode = GAIN_MODE_MANUAL;
config.gain = 30.0;                  // 30dB
config.buffer_size = 4 * 1024 * 1024; 

// Set sampling parameters
params.num_samples = config.sampling_rate; 
params.buffer_size = config.buffer_size;
params.buffer = malloc(params.num_samples * 4); 

// start receive samples
ret = device_receive_samples(dev, &params);
if (ret < 0) {
    LOG_ERROR("Failed to receive samples: %d", ret);
    free(params.buffer);
    return;
}
```

```c
// Receive samples
int device_receive_samples(struct device *dev, struct device_sample_params *params)
{
    if (!dev || !params || !params->buffer) {
        return -ERR_INVALID_PARAM;
    }

    int loop_count = 0;

    // Check device status
    if (dev->state != DEVICE_STATE_RUNNING) {
        device_set_error(dev, ERR_INVALID_STATE, "Device not in running state");
        return -ERR_INVALID_STATE;
    }

    // Initialize parameters
    params->bytes_received = 0;
    params->overflow = false;
    size_t total_bytes = params->num_samples * 4; // Each sample is 4 bytes (2 bytes for I and 2 bytes for Q)
    // params->num_samples = 61.44e6
    // params->buffer_size = 4 * 1024 * 1024 (4MB)
    
    // Loop to read until enough samples are obtained
    while (params->bytes_received < total_bytes) {
        size_t remaining = total_bytes - params->bytes_received;
        size_t to_read = (remaining < params->buffer_size) ? 
                        remaining : params->buffer_size;

        // Read data
        int ret = dev->ops->read(dev, 
                               (uint8_t*)params->buffer + params->bytes_received, 
                               to_read);
        
        if (ret < 0) {
            device_set_error(dev, ERR_DEVICE_IO, "Read failed");
            return ret;
        }
        
        if (ret == 0) {
            // No more data to read
            break;
        }

        params->bytes_received += ret;

        loop_count++;
    }

    // Check if enough samples have been collected
    if (params->bytes_received < total_bytes) {
        LOG_ERROR("Failed to receive requested number of samples: %zu/%zu bytes", 
                 params->bytes_received, total_bytes);
        return ERR_BUFFER_UNDERFLOW;
    }

    LOG_INFO(" ==== Received count: %d, total_bytes: %zu, buffer_size: %zu\n", 
           loop_count, total_bytes, params->buffer_size);

    return 0;
}
```

```c
// Read data
// dev->ops->read func
static ssize_t read(struct device *dev, void *buf, size_t len)
{
    struct ad9361_device *ad9361 = container_of(dev, struct ad9361_device, base);
    if (!ad9361->rxbuf) {
        LOG_ERROR("RX buffer not created");
        return -1;
    }

    struct timespec start, end;
    // Record start time
    clock_gettime(CLOCK_MONOTONIC, &start);

    // Wait for new data
    ssize_t nbytes = iio_buffer_refill(ad9361->rxbuf);
    
    // Record refill time
    clock_gettime(CLOCK_MONOTONIC, &end);
    double refill_time = (end.tv_sec - start.tv_sec) + 
                        (end.tv_nsec - start.tv_nsec) / 1e9;
    
    if (nbytes < 0) {
        LOG_ERROR("Buffer refill failed: %zd", nbytes);
        return nbytes;
    }

    // Get data pointer
    void *data_start = iio_buffer_start(ad9361->rxbuf);
    void *data_end = iio_buffer_end(ad9361->rxbuf);
    ptrdiff_t p_inc = iio_buffer_step(ad9361->rxbuf);
    size_t samples_in_buffer = (data_end - data_start) / p_inc;

    // Limit copy size
    size_t bytes_to_copy = samples_in_buffer * sizeof(int16_t) * 2;  // I/Q pairs
    if (bytes_to_copy > len) {
        bytes_to_copy = len;
    }

    // Record copy start time
    clock_gettime(CLOCK_MONOTONIC, &start);

    memcpy(buf, data_start, bytes_to_copy);
    
    // Record copy end time
    clock_gettime(CLOCK_MONOTONIC, &end);
    double copy_time = (end.tv_sec - start.tv_sec) + 
                      (end.tv_nsec - start.tv_nsec) / 1e9;
    LOG_DEBUG("Read stats: refill=%.6fs, copy=%.6fs, bytes=%zu", 
              refill_time, copy_time, bytes_to_copy);
    
    return bytes_to_copy;
}
```

```
// Device operation implementation
static int ad9361_open(struct device *dev)
{
    struct ad9361_device *ad9361 = container_of(dev, struct ad9361_device, base);
    
    // Initialize IIO context
    ad9361->ctx = iio_create_default_context();
    if (!ad9361->ctx) {
        device_set_error(dev, ERR_DEVICE_INIT, "Failed to create IIO context");
        return -1;
    }
    
    // Print all available devices for debugging
    unsigned int nb_devices = iio_context_get_devices_count(ad9361->ctx);
    LOG_DEBUG("Found %u IIO devices", nb_devices);
    
    for (unsigned int i = 0; i < nb_devices; i++) {
        const struct iio_device *device = iio_context_get_device(ad9361->ctx, i);
        LOG_DEBUG("Device %u: %s", i, iio_device_get_name(device));
    }
    
    // First find the PHY device for configuration
    ad9361->phy = iio_context_find_device(ad9361->ctx, "ad9361-phy");
    if (!ad9361->phy) {
        device_set_error(dev, ERR_DEVICE_NOT_FOUND, "AD936x PHY device not found");
        goto err_destroy_context;
    }
    LOG_INFO("Found AD936x PHY device");

    // Then find the data device
    ad9361->data_dev = iio_context_find_device(ad9361->ctx, "cf-ad9361-lpc");
    if (!ad9361->data_dev) {
        device_set_error(dev, ERR_DEVICE_NOT_FOUND, "AD936x data device not found");
        goto err_destroy_context;
    }
    LOG_INFO("Found AD936x data device");
    
    // Find I/Q channels on the data device
    ad9361->rx_i = iio_device_find_channel(ad9361->data_dev, "voltage0", 0);  // Use 0 instead of false
    if (!ad9361->rx_i) {
        LOG_ERROR("Failed to find I channel");
        goto err_destroy_context;
    }
    LOG_INFO("Found I channel");

    ad9361->rx_q = iio_device_find_channel(ad9361->data_dev, "voltage1", 0);  // Use 0 instead of false
    if (!ad9361->rx_q) {
        LOG_ERROR("Failed to find Q channel");
        goto err_destroy_context;
    }
    LOG_INFO("Found Q channel");

    // Check available channels
    unsigned int num_channels = iio_device_get_channels_count(ad9361->data_dev);
    LOG_INFO("Available channels: %u", num_channels);

    // Transition device state
    return device_transition_state(dev, DEVICE_STATE_OPENED);

err_destroy_context:
    iio_context_destroy(ad9361->ctx);
    return -1;
}
```

```c
// start device function
// rx
static int start(struct device *dev)
{
    struct ad9361_device *ad9361 = container_of(dev, struct ad9361_device, base);
    LOG_DEBUG("Starting AD936x device");

    // 1. Check current state
    if (dev->state != DEVICE_STATE_CONFIGURED) {
        LOG_ERROR("Invalid state for start: %d", dev->state);
        return ERR_INVALID_STATE;
    }

    // 2. Enable channels
    iio_channel_enable(ad9361->rx_i);
    iio_channel_enable(ad9361->rx_q);

    // 3. Set buffer parameters
    int ret = iio_device_set_kernel_buffers_count(ad9361->data_dev, 8);
    if (ret < 0) {
        LOG_ERROR("Failed to set kernel buffers count");
        goto error_disable_channels;
    }

    // 4. Create receive buffer
    ad9361->rxbuf = iio_device_create_buffer(ad9361->data_dev, 
                                            1024 * 1024,
                                            false);
    if (!ad9361->rxbuf) {
        LOG_ERROR("Failed to create buffer");
        goto error_disable_channels;
    }

    // 5. Wait for device stabilization
    usleep(500000);
    
    return device_transition_state(dev, DEVICE_STATE_RUNNING);

error_disable_channels:
    iio_channel_disable(ad9361->rx_i);
    iio_channel_disable(ad9361->rx_q);
    return ERR_DEVICE_CONFIG;
}
```

dNechita, 2025-03-21:
Hi,

I assume you are running this directly on the embedded device, right?
What periods of time are you getting for the execution of memcpy?

whale1st, 2025-03-21:
> Hi,
> 
> I assume you are running this directly on the embedded device, right? What periods of time are you getting for the execution of memcpy?
> 
> If you comment out the memcpy (just for testing purposes) how much does the time exection of device_receive_samples() decreases compared to 1.4 s you mentioned?

Hi Dan,

Thank you so much for your questions and for taking the time to look into this issue—I really appreciate your help and interest!

Yes, I’m running the code directly on the embedded device.   Here are the memory usage details for reference:

```shell
Mem: 334136K used, 173396K free, 76K shrd, 0K buff, 46860K cached
CPU:  12% usr  17% sys   0% nic  68% idle   0% io   0% irq   2% sirq
Load average: 0.27 0.12 0.04 2/63 1964
PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
1651     1 root     S     354m  71%  31% ./recv-samples
1946  1675 root     R     2452   0%   0% top
```
And here’s the output of free -h:
```shell
total        used        free      shared  buff/cache   available
Mem:         495.6M      277.3M      169.4M       76.0K       48.9M      166.9M
Swap:             0           0           0
```

Regarding the execution time of memcpy, I measured it and found that each call takes approximately 0.025 seconds to copy 1M sample points.  When I comment out memcpy, the execution time of device_receive_samples() decreases significantly—from around 1.4 seconds to approximately 0.9 seconds.  This suggests that memcpy is contributing a notable portion of the overall execution time, especially given the volume of data being processed.

If there’s anything else you’d like me to test or clarify, please don’t hesitate to let me know. I’d be happy to provide more details or run additional tests to help you better understand or resolve this issue.

closed 2025-04-18 by whale1st