analogdevicesinc/libiio #1270: find_device raises error if null starting from d376e5668881e4fc2727bb4a33083820e27069a8

gastmaier, 2025-04-08:
Starting from https://github.com/analogdevicesinc/libiio/commit/d376e5668881e4fc2727bb4a33083820e27069a8
[_find_device.errcheck = _check_null](https://github.com/analogdevicesinc/libiio/commit/d376e5668881e4fc2727bb4a33083820e27069a8#diff-2ef8fa9f650e418c7327fd6dd9076b8fd965b0115b8495a065d6e51cb02c8c9eR343),
but logic at pyadi-iio wasn't updated, causing logic like

```
        self._rx_dsa = self._ctx.find_device("hmc425a")
        if not self._rx_dsa:
            self._rx_dsa = self._ctx.find_device("hmc540s")
```
to fail
```
Traceback (most recent call last):
  File "/mnt/wsl/data/repos/pyadi-iio/examples/QuadMxFE_dual_example.py", line 85, in <module>
    multi = adi.QuadMxFE_multi(primary, [secondary], primary_jesd, [secondary_jesd])
  File "/mnt/wsl/data/repos/pyadi-iio/adi/QuadMxFE_multi.py", line 49, in __init__
    self.primary = QuadMxFE(uri=primary_uri)
                   ~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/mnt/wsl/data/repos/pyadi-iio/adi/ad9081_mc.py", line 329, in __init__
    self._clock_chip = self._ctx.find_device("hmc425a")
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/iio.py", line 1480, in find_device
    dev = _find_device(self._context, name_or_id_or_label.encode("ascii"))
  File "/usr/lib/python3.12/site-packages/iio.py", line 61, in _check_null
    raise OSError(err, _strerror(err))
OSError: [Errno 0] Success
```

The intent of this issue is to decide if this line needs to be reverted, or pyadi-iio code needs to be update to try-catch.

dNechita, 2025-04-08:
I have recently made a fix (https://github.com/analogdevicesinc/libiio/pull/1267) that should solve this issue. Can you check with latest main? Thanks!

gastmaier, 2025-04-08:
Nice, thanks, yes this resolves it.

closed 2025-04-08 by gastmaier