analogdevicesinc/libiio #1276: Default installs into /usr and not /usr/local

dlmiles, 2025-04-21:
I've had to manually tweezer out and delete all the files it has installed into the system areas.  How rude of it!

Maybe this project should at least output the install location that is configured, so the user is not seeing it after `sudo`, when it has configured by default to use /usr:
```
cmake ..
...
NOTICE: -DCMAKE_INSTALL_PREFIX=/usr is set by default, if you expect /usr/local please configure cmake
```

The alternative of not being a good citizen is to have to test build and test make install into a docker container and examine the snapshot before/after to confirm it only touches

dlmiles, 2025-04-22:
Thanks for the reply, I'll see what I can do re patch.

Thanks for saying `uninstall` is working but when `sudo make install` did something unexpected, excuse me if I don't want it to run `sudo make uninstall` (ask the project to do something destructive) trust was lost already, project needs a fuller audit to continue, because those choices indicate whomever created it has a different view of the world and that may cause problems in other areas.

I don't think the cmake output is super-noisy, how many users are expected to read and understand every line output ?  those that do look, when they are looking, are they searching for specific information?  data that is not present can not be found.
Having run 1000s of CI noisy output is good, it saves time, less time spent on diagnostic because information is already there to see, no need to search/find-out-to-see/re-run-CI all takes a lot of time to fix a configuration/build issue.  When an immediate fix can often be made because the information is present in output.

rgetz, 2025-04-27:
Just as a note - the [README_BUILD.md](https://github.com/analogdevicesinc/libiio/blob/main/README_BUILD.md) already includes the note:

Cmake Options          | Default | Target | Description                                    |
---------------------- | ------- | -------| ---------------------------------------------- |
`CMAKE_INSTALL_PREFIX` | `/usr` |   Linux | default install path |

closed 2025-04-29 by dNechita

dlmiles, 2025-04-30:
Thanks for resolving this matter.  Making it clear at the end of cmake configure output would have been enough to prevent any surprise.

commit fe11ae86 (2025-04-29): cmake/doc: Ensure users see the default install path

Resolves #1276

document make uninstall in the README_BUILD.md

during configure - print out the install path, and a hint on how to
change it if people want.