analogdevicesinc/libiio #1277: Ship the udev rules upstream

AsciiWolf, 2025-04-21:
Please, consider submitting the [udev rules](https://github.com/analogdevicesinc/libiio/blob/main/libiio.rules.cmakein) to upstream (systemd). Here is an example how to do it: https://github.com/systemd/systemd/pull/20561

Without this, it is [not possible](https://github.com/flatpak/xdg-desktop-portal/issues/227#issuecomment-946760482) to use PlutoSDR devices in Flatpak and other special configurations where it is not possible to install the libiio libs/udev rules system-wide.

Please see the upstream ticket: https://github.com/systemd/systemd/issues/37019

rgetz, 2025-04-22:
It's normally part of the upstream distribution developers that package libiio to include the udev rules in their packages...

which distribution are you using?

AsciiWolf, 2025-04-22:
Fedora Silverblue. Please read the linked upstream tickets for more details.

(It is not limited only to immutable distributions, but all Flatpak/AppImage apps.)

rgetz, 2025-04-23:
while libiio is used for Pluto - libiio is much more generic, and there are not device specific rules inside libiio today - thus the udev rules that we use in libiio are likely not appropriate for upstream systemd. (since they use libiio utilities, which may not be installed).

The best place to ask this sort of question might be in the plutosdr repo. https://github.com/analogdevicesinc/plutosdr-fw