analogdevicesinc/libiio #1278: libiio-v1: How to write data to a kfifo-backed IIO buffer?

schwesinger, 2025-04-24:
Hi,

in libiio-v1 I'm looking for a way to write data to a kfifo-backed output buffer. IIUC, in the new libiio API `iio_block`s are used to actually exchange data so my approach looked like this:

1. Set up local context
2. Get reference to device
3. Get reference to channel (and create channels mask)
4. Create an IIO buffer
5. Create an IIO block
6. Use `iio_channel_write()` to write user-supplied data into the block

However, `iio_buffer_create_block()` (to be more precise `local_create_block()`) always fails for kfifo-backed buffers because they neither provide the DMABUF nor the high-speed MMAP interface.

So, how can I programmatically write data to a kfifo-backed IIO buffer with the new API?

schwesinger, 2025-04-25:
The problem I faced was two-fold: my first approach as outlined above was not quite right. Instead of using `iio_channel_write()`, the `iio_block_*()` functions had to be used to write data into the `iio_block` and to actually enqueue it (see https://github.com/analogdevicesinc/libiio/wiki/libiio_0_to_1#accessing-data).

However, I think `iio_buffer_create_block()` currently has a bug that prevents creating `iio_block`s for the kfifo (and MMAP!) use-cases. See https://github.com/schwesinger/libiio/commit/8ecbf69c10ea7fca19d58e302bbd756fb30f6ea0 for more details and a patch.

With this patch and the updated approach, I was able to successfully write data to a kfifo-backed buffer.

Please let me know, if the patch is suitable for inclusion into the project, I can create a PR then.

rgetz, 2025-04-25:
This makes sense to me (thanks for the well written patch).

Maybe @pcercuei or @nunojsa or @dNechita can have a look at the details.

Thanks
-Robin

pcercuei, 2025-04-25:
The patch looks good to me, you can PR it.

schwesinger, 2025-04-28:
I think this is resolved now. Thank you to the maintainers for handling this so quickly, much appreciated!